cesanta/mongoose-os #7: Wifi.status() gives garbage on first run

Spritetm, 2015-07-24:
I'm testing your smart.js code, and I think I found a bug in it. When I first started it up, I tried to get the WiFi status:

smartjs 12040/3456$ Wifi.status()
"�W�?"

I compiled this with the 1.2.0 SDK, and it looks like the bug originally stems from there: when the WiFi station isn't online for whatever reason, wifi_station_get_connect_status() returns -1 (well, it's an u8 so actually 255.) I'll see if I can get that fixed or documented in the SDK, but in general, maybe the select statement in v7_esp.c, Wifi_status could use a default case so it doesn't return uninitialized memory when the WiFi status value is something weird?

gelraen, 2015-07-24:
Good point, I'll fix that.

gelraen, 2015-07-24:
Fixed in https://github.com/cesanta/smart.js/commit/0f8516cdd2f885ebd4ad554d180cc17825e81a68

closed 2015-07-24 by gelraen