cesanta/mongoose-os #12: Roadmap

boneskull, 2015-08-05:
This really interests me, as my least favorite part of working with these devices is writing C. 

I'm wondering what the roadmap looks like?  I have a few things off the top of my head that I'd like to see:

- CoAP client
- CoAP server
- decouple the cloud functionality from the core?  If I pick up Smart.js, I won't use/need Cesanta's servers. 

I'm also curious how well the custom JS engine will withstand the test of time. it's been tried before with things like Espruino, and the result was/is pretty painful. I think part of the issue in that case was a non-JS developer writing a JS API. `Object.emit()`? the Tessel team threw up their hands at this nonsense and decided to just run Linux.  The other issue of course is that JS developers come from the browser and NodeJS and expect their favorite functions to be there. What might be a cool idea would be to transpile from everyday JS to Smart.js-speak. 

Keep hacking!  This is exciting stuff.

cpq, 2015-08-05:
We plan to re-use Fossa (https://github.com/cesanta/fossa) for networking, which will add a huge bunch of networking, including coap. It's just ESP's SDK doesn't provide BSD sockets API, so it was easier to put together a custom HTTP client/server based on their native API.

With existing HTTP API, you can already use other clouds. And yes, we plan to move hardcoded "cesanta.com" to the config, to let people to use their own servers.

W.r.t. JS API: if you have a specific suggestions, we'd be more than happy to hear.

cpq, 2015-08-07:
Closing this.
Please reopen if you have suggestions w.r.t. API.

closed 2015-08-07 by cpq

boneskull, 2015-08-12:
I'll let you know. Haven't played with it yet.