cesanta/mongoose-os #13: File.list returns same results

goliatone, 2015-08-08:
`File.list` returns always the same list:

```js
File.list('/var')
```
>"gpio.js","I2C.js","init.js","MC24FC.js","MC24FC_test.js","MCP9808.js","MCP9808_test.js","MPL115A1.js","smart.js","sys.json","user.json"]

```js
File.list('/etc/')
```

>"gpio.js","I2C.js","init.js","MC24FC.js","MC24FC_test.js","MCP9808.js","MCP9808_test.js","MPL115A1.js","smart.js","sys.json","user.json"]

cpq, 2015-08-09:
Is it on ESP8266 ?
SPIFFS is a flat file system, and doesn't support directories. That might be the reason.
Please reopen if that's not on SPIFFS.

closed 2015-08-09 by cpq

goliatone, 2015-08-09:
@cpq Look at the quick start guide on the repo's README.

>6. Type some JavaScript expression to the console and press enter. Smart.js evaluates the expression and prints evaluation result: 

[image]

goliatone, 2015-08-09:
@cpq Also, why would you close this issue before getting a response?

cpq, 2015-08-10:
@goliatone the issue is resolved: SPIFFS is flat.
Please reopen if you think it's not resolved.