cesanta/mongoose-os #15: Feature request: ADC.read && setTimeout

Bravo13, 2015-08-09:
So, is it already realized in smart.js to read data from the adc and what about js functions setTimeout and setInterval?

mmikulicic, 2015-08-10:
We do have a currently very basic support for `setTimeout`. It doesn't yet support timer cancellation and on the ESP8266 platform only one timer can be be used at a time (calling setTimeout again will reset the previously set timer).

We'll definitely allow concurrent timers soon, and very likely also implement cancellation support and `setInterval` as well.

The ADC interface is still TODO.

eallik, 2015-11-23:
+1 any news on this? I actually had a transducer connected to my NodeMCU with the voltage range down to [0.0, 1.0] and all, and then realized I couldn't read it from JavaScript... :(

mmikulicic, 2015-11-23:
Simple ADC API implemented.

It currently works only on ESP8266, which has only one ADC. Usage:

```js
ADC.read(0)
```

returns a JS number in the `[0.0, 1.0]` range.

I tried it only by connecting to ground and VCC. I got .0029296875 and 1.0 respectively
Each device might have to be calibrated. We are just issuing a call the the ESP8266 SDK which offers a very basic interface to the ADC.

closed 2015-11-23 by mmikulicic

eallik, 2015-11-23:
This is awesome; I think exactly what one would expect. Going to be using it in a few min. Thanks!