cesanta/mongoose-os #16: Docker container reference seems to be broken in 'make.sh'

ddm, 2015-08-18:
In the `platforms/esp8266/make.sh` file, the referenced docker container is:

    docker.cesanta.com:5000/esp8266-build-oss:1.3.0-cesanta

It seems that the corresponding docker repository is empty:

    Unable to find image 'docker.cesanta.com:5000/esp8266-build-oss:1.3.0-cesanta' locally
    Pulling repository docker.cesanta.com:5000/esp8266-build-oss
    Error: image esp8266-build-oss:1.3.0-cesanta not found

Changing the `make.sh`file seems to fix the issue:

    SDK=cesanta/esp8266-build-oss

Is the container located at https://hub.docker.com/r/cesanta/esp8266-build-oss/ still valid for build?

mmikulicic, 2015-08-18:
The image in the docker hub repo has a quite old SDK and it's not compatible with latest smartjs. 

Thanks for reporting, I'll take a look

mmikulicic, 2015-08-19:
which version of docker do you use?

ddm, 2015-08-20:
Docker version 1.8.1, build d12ea79

Dimitri del Marmol

dimitri.delmarmol@gmail.com
+32 485 88 49 40

mmikulicic, 2015-08-20:
Issue caused by https://github.com/docker/docker/issues/15640 affecting https://github.com/cesanta/docker_auth/issues/22

In the meantime I synced the image to cesanta/esp8266-build-oss:1.3.0-cesanta

ddm, 2015-08-20:
Perfect, I modified `make.sh` and it works

    SDK=cesanta/esp8266-build-oss:1.3.0-cesanta

rojer, 2015-08-20:
this should be fixed now with cesanta/docker_auth#23

closed 2015-08-20 by rojer