cesanta/mongoose-os #19: Doesn't connect to AP

ybouhjira, 2015-08-23:
```js
print('Hello world');

Wifi.setup('Dlink', 'password');

Wifi.changed(function(status) {
  print('Wifi status changed to ', status);
  if(status == 4) {
    Http.get("http://jsonip.com", function(data, error) {
      if (error) {
        print("error ", error);
      } else {
        print("my ip is ", JSON.parse(data).ip);
      }
    });
  }
});
```

I'm trying to connect to my WPA2 router but the status never reaches 4.

mmikulicic, 2015-08-23:
Could you please enable debug output with Mode.debug(1) and send the whole output here?

ybouhjira, 2015-08-23:
```

    sl��r�blc��no�lon���cp��lslrlp�o��lbn�|�ll�c��on��onl`or���n�l`p�o�r�����bn�|����c��on�r�����l�bn�|ll��b��on���o��lrl
My Cesanta cloud ID: //api.cesanta.com/d/_2FBL7M 
My PSK: 5sX8MdBjYQ 
Use these credentials to add this device on https://dashboard.cesanta.com/ 
Scanning nets 

Executing init.js
Hello world 
scandone
wifi scan failed: 1
Cannot find known network, use Wifi.setup 
f 0, init.js called

smartjs 24664/2004$ scandone
add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with Dlink, channel 11
Wifi status changed to 
dhcp client start...
Wifi status changed to 3 
ip:192.168.0.50,mask:255.255.255.0,gw:192.168.0.1
```

mmikulicic, 2015-08-23:
Our documentation is probably misleading.

```
Wifi.changed(cb) - invokes cb whenever the connection status changes:
0: connected
1: disconnected
2: authmode changed
3: got ip
4: client connected to ap
5: client disconnected from ap
```

"Client connect to ap" is received when another host connects to the ESP when it's in Access Point mode (or the hybrid stationAP mode).

You are interested to the "3: got ip" event, which signals that the ESP itself not only connected to the API but also received a DHCP response with an IP address. This happens after "0: connected" which is received when the ESP connects to an access point.

I guess we can improve the doc. In the master branch we refactored the event codes here in order to make the WiFi API cross platform.

mmikulicic, 2015-08-24:
@ybouhjira, did I address your issue ?
Does it work otherwise?

alashkin, 2016-05-13:
This code received a lot of fixes last time, documentation updated as well.
See here: https://github.com/cesanta/mongoose-iot/blob/master/docs/javascript-api/wifi.md
```
Wifi.changed(cb): Invokes cb whenever the connection status changes:
   Wifi.CONNECTED: connected
   Wifi.DISCONNECTED: disconnected
   Wifi.GOTIP: got ip
```
Should be ok now.

closed 2016-05-13 by alashkin