cesanta/mongoose-os #20: Http.get with master build

necrodome, 2015-09-07:


mmikulicic, 2015-09-08:
Hi @necrodome 

1. the "http://96.126.98.124" version doesn't work because we didn't implement a literal IP address parser. It will work once we manage to port mongoose to ESP (currently blocked on a few SDK issues). DNS names seemed enough for a quick proof of concept, but I guess I could just add it.

2. I just checked and `http://jsonip.com` it works for me. Can you ping your esp? (You can see your local ip with `Wifi.ip()`). If you connect on the same wifi, what DNS server does your computer get?

mmikulicic, 2015-09-08:
FYI http://xip.io/

necrodome, 2015-09-08:
Thanks for the pointer. I tried with a a different network, and it managed to make the request.

Is there a way to tell smart.js to use a specific dns server?

mmikulicic, 2015-09-08:
Not from JS currently.
You'd have to invoke `espconn_dns_setserver` (Espressif SDK) in the C code and rebuild the firmware.

I think it might make sense to expose it via JS.

mmikulicic, 2015-09-08:
Hm, ESP doc states:

```
Only if ESP8266 DHCP client is disabled (wifi_station_dhcpc_stop), this API
can be used
```

So yes, it would be possible but it comes with a price.

Our goal is to integrate mongoose (github.com/cesanta/mongoose) for async networking IO. It includes an async DNS resolver so we would have more freedom setting the DNS.

necrodome, 2015-09-08:
Great. Do you have a timeframe or list of issues for mongoose integration?

mmikulicic, 2015-09-08:
No ETA; ESP mongoose integration is blocked on technical issues.
For now we just try to do our best to make it useful with reduced functionality. Custom DNS servers are not top priority at the moment.

Other supported platforms (e.g. TI  CC3200) already use mongoose for networking.

You might also want to try the "posix" port and run it on your laptop.

necrodome, 2015-09-08:
Thanks for all the info.

closed 2015-09-08 by necrodome