cesanta/mongoose-os #22: Compilation instructions

pastukhov, 2015-09-24:
with ```sh make.sh``` i'm getting ```Makefile:18: *** Use make.sh to build project within docker container. To build against locally installed ESP SDK define SDK_FLAVOUR (with RTOS or OSS value) enviroment variable.  Stop.```
with ```sh make.sh  RTOS``` : 
```
Unable to find image 'docker.cesanta.com:5000/esp8266-build-rtos:1.2.0-cesanta-r4' locally
1.2.0-cesanta-r4: Pulling from docker.cesanta.com:5000/esp8266-build-rtos
a8c34f0f9db9: Pull complete 
46745474f5aa: Pull complete 
7163ab7a738c: Pull complete 
3885a86c7f89: Pull complete 
fee9f8ef6b54: Pull complete 
00603c7e53a4: Already exists 
6071b4945dcf: Already exists 
5bff21ba5409: Already exists 
e5855facec0b: Already exists 
8251da35e7a7: Already exists 
09415d9209b6: Already exists 
039ea637f66d: Already exists 
56628a10fdaa: Already exists 
95fd921a6c96: Already exists 
82e9bde453bb: Already exists 
0c2a69e609b2: Already exists 
0f277c74a47a: Already exists 
a53cf678171e: Already exists 
f24ef6cf7338: Already exists 
5ea338a5df36: Already exists 
Digest: sha256:53f1eb6cd4829df09b0918158bf11f18cedd3542d86c8f0e050ca1af3a3c5824
Status: Downloaded newer image for docker.cesanta.com:5000/esp8266-build-rtos:1.2.0-cesanta-r4
  CC user/c_sscanf.c -> build/c_sscanf.o
  CC user/dht11.c -> build/dht11.o
  CC user/disp_task.c -> build/disp_task.o
  CC user/disp_task_net_test.c -> build/disp_task_net_test.o
  CC user/esp_coredump.c -> build/esp_coredump.o
user/esp_coredump.c:14:20: fatal error: base64.h: No such file or directory
 #include "base64.h"
                    ^
compilation terminated.
make: *** [build/esp_coredump.o] Error 1

maxogden, 2015-09-26:
I get a similar, but different, error when running the same thing (on ubuntu 14.04 x64)

```
max@burrito:~/smart.js/platforms/esp8266$ sudo sh make.sh
Unable to find image 'docker.cesanta.com:5000/esp8266-build-oss:1.3.0-cesanta' locally
Pulling repository docker.cesanta.com:5000/esp8266-build-oss
2015/09/25 17:51:06 HTTP code: 404
```

mmikulicic, 2015-09-26:
@maxogden by any chance, did you try to `docker login docker.cesanta.com:5000` ?

Could you try to run?

```sh
grep docker.cesanta.com ~/.docker/config.json ~/.dockercfg
```

mmikulicic, 2015-09-26:
OSS build works, RTOS still broken:

```
make: *** No rule to make target `mongoose.c', needed by `build/mongoose.o'.  Stop.
```

cpq, 2015-09-27:
https://github.com/cesanta/smart.js/commit/cc18d18a6907b03fbbf8f0552925ee666dc3162b fixed this.
Thanks @mmikulicic , @pastukhov , @maxogden

closed 2015-09-27 by cpq