cesanta/mongoose-os #25: Flashnchips: "Failed to merge flash filesystem"

fedy2, 2015-09-30:
I've flashed my ESP8266 01 a first time successfully but I got no responses from the live console.
I've decided to flash it a second time but I get the following error message "Failed to merge flash filesystem". 
What I can do about?

I'm using an ESP8266 01, connected trough a FT232 adapter. Flash'n'Chips is running under Windows 7 x64.

mmikulicic, 2015-09-30:
CC @gelraen

mmikulicic, 2015-09-30:
@fedy2 I don't know what is the root cause in your situation, but let's start by disabling the merge FS feature and get your device working:

Invoke flash'n'chips from cmdline and pass `--overwrite-flash-fs --gui` params

Not sure if `--gui` is necessary on windows, I have a vague memory of windows apps having to be either console or gui (defined at link time). @gelraen knows more about it.

gelraen, 2015-09-30:
Yes, Windows binary doesn't print anything in the console :(

@fedy2 , can you also add `--log fnc.log --V=4` and share `fnc.log` somehow? This may help to figure out why exactly merging fails.

fedy2, 2015-09-30:
Tested with logs enabled: http://pastebin.com/Tc90bLwW

fedy2, 2015-09-30:
With the --overwrite-flash-fs param the flash operation completed successfully.
However the device don't reply to console commands. I will open a new issue about it.

gelraen, 2015-09-30:
```WARNING: Current flash params bytes: 0240```

This doesn't sound right for ESP-01. It should be 0000. You can try to override it with `--esp8266-flash-params 0x0000`

fedy2, 2015-09-30:
With the `--esp8266-flash-params 0x0000` param the console works after a re-connection.
However the first message after flashing is un-readable:
```
--- flashed successfully
--- connected
load 0x40100000, len 31868, room 16 
�ƄE�慆ąb愄Ć����DÇ�ĥ����������Ǆ�b���bŧĄ�DÇ�ĥ��Ć���Ĥ熄b�ƄE�慆ĥb�Ą�E�
--- disconnected
```

gelraen, 2015-09-30:
Can you try to disconnect GPIO0 from GND and reset the board?

fedy2, 2015-09-30:
After the reset I've obtained the device id and psk. Thank you.

mmikulicic, 2015-09-30:
Does it work now without the overwrite fs and/or flash params options?

fedy2, 2015-09-30:
Nope. I need to specify both the parameters.

alashkin, 2016-05-20:
FNC received a lot of fixes, not sure this issue is actual (I have not seen this problem for a long time).
I'm closing issue, but feel free to reopen.

closed 2016-05-20 by alashkin