cesanta/mongoose-os #27: flashnchips: unknown chars after ESP flash

fedy2, 2015-09-30:
After the ESP8266 01 has been flashed I get this message in console:
```
--- flashed successfully
--- connected
load 0x40100000, len 31868, room 16 
??E??ab?C????DÇ?h????????????b???btA?DÇ?h??C???H?b??E??hb?A?E?????CrDevice id: //api.cesanta.com/d/XXXXXX
Device psk: XXXXXXXX 
Cloud: //api.cesanta.com 
```

I've obfuscated the id and the psk.

mmikulicic, 2015-09-30:
This is the output of esp SDK spat out at a baudrate that many TTL uart adapters do not support, hence we decided to default to 115200 baud.

We currently do not know how to inhibit the output so early at boot. There might be something output by the rom itself.

cpq, 2016-12-27:
Closing as obsolete

closed 2016-12-27 by cpq