cesanta/mongoose-os #28: No data for registered device

fedy2, 2015-09-30:
As stated in the Start Guide (https://github.com/cesanta/smart.js/tree/master#quick-start-guide) I've registered the device after the wi-fi configuration but I don't see any data chart visualized.
I've tried also to open the dashboard IDE but the device status is offline.

The wi fi configuration seems working, I see the device as result of a network scan.
The device is an ESP8266 01 flashed with ALPHA 2 firmware.

fedy2, 2015-09-30:
Here some output from the console:
```
smartjs 30048/2484$ Wifi.status()
got ip
smartjs 29464/2388$ Wifi.ip()
192.168.1.108
smartjs 29464/2388$
```

mmikulicic, 2015-10-01:
Thank you, we broke the dashboard visualisation. 

Could you check if you an see your raw data on https://dashboard.cesanta.com/#/db by executing this query:

`{src=~".*"}`

?

fedy2, 2015-10-01:
I get this result:
```
[[{"__name__":"temperature","sensor":"1","src":"//api.cesanta.com/d/dev_84227"},[[1443690822306,37.5488386757]]],[{"__name__":"temperature","src":"//api.cesanta.com/d/fakedevice03"},[[1443628202572,27.03]]],[{"__name__":"temperature","src":"//api.cesanta.com/d/fakedevice01"},[[1443628201987,13.94]]],[{"__name__":"temperature","src":"//api.cesanta.com/d/fakedevice02"},[[1443628202260,19.3]]]]
```

I don't remember if dev_84227 is an old device. Anyway I don't see data from the last device.

cpq, 2015-10-01:
@fedy2 I've restored the visualization part, could you try again please?

fedy2, 2015-10-01:
I see the chart scrolling on the left but no values on it. In the IDE area the device result offline.

mmikulicic, 2015-10-01:
@fedy2, we currently have a bug with multiple tabs/windows. Could you see if you have the same problem when you have only one tab/window open on dashboard.cesanta.com ?

fedy2, 2015-10-01:
@mmikulicic I'm using only one browser tab/windows with one "instance" of the dashboard. When I've used the tab term I've referred to the different areas of the dashboard.

mmikulicic, 2015-10-01:
ok, we fixed that, unrelated, bug anyway

gelraen, 2015-10-01:
@fedy2 what is the ID your device is using now?

fedy2, 2015-10-01:
ID: //api.cesanta.com/d/dev_65156

gelraen, 2015-10-01:
From what I see in the logs your device never connected and all commands sent to it ended up in the queue.

fedy2, 2015-10-01:
I see it in my network. There is a way to test if it can communicate with your servers? Any console command?

fedy2, 2015-10-01:
I've run this code:
```
Http.get("http://jsonip.com", function(data, error) {
  if (error) {
    print("error ", error);
  } else {
    print("my ip is ", JSON.parse(data).ip);
  }
});
```

And I got the IP printed:
```
smartjs 12704/1812$ my ip is  87.XX.XX.195 
```

So I think the device can communicate with the world.

gelraen, 2015-10-01:
And you ran `demo()` function, right?

fedy2, 2015-10-02:
Nope :( forgot about. 
I've tried now to run it different times but the ESP seems hanging.
The last time I got this:
```
smartjs 16912/1980$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset
```

Can be a power issue?

mmikulicic, 2015-10-02:
Please turn on debug logs with `Debug.mode(1)` and try again

fedy2, 2015-10-02:
Here the output:
```
demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
Pushing metric 

 ets Jan  8 2013,rst cause:4, boot mode:(3,0)

wdt reset
load 0x40100000, len 26380, room 16 
tail 12
chksum 0xd1
ho 0 tail 12 room 4
load 0x3ffe8000, len 2564, room 12 
tail 8
chksum 0x1c
load 0x3ffe8a10, len 6320, room 0 
tail 0
chksum 0xbe
csum 0xbe
���	yr�'���θ���������p�b�#�n�$��b��>��rn�$���r��n���rn��N����������8�b�#�n�$p���r�'bl���;�	n�{��l���	�b��ؒ��`��#��	yr�'���θ���������8b���nn�$�b|��rn�$�Ìr��������rn�ď���l�r�n�����rn��θ��������l��r��?b��r���Ă���p`����8��Ē`Ĝ�r����n������Nǲ�mode : softAP(1a:fe:34:9c:a2:a1)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100
apnetif->dhcps_pcb = 1073709328
bcn 0
del if1
usl
sul 0 0
mode : sta(18:fe:34:9c:a2:a1)
add if0
 
Device id: //api.cesanta.com/d/dev_76813 
Device psk: psk_17496 
Cloud: //api.cesanta.com 
HELO! Type some JS. See https://github.com/cesanta/smart.js for more info. 

smartjs 15128/1968$ 
```

alashkin, 2016-05-13:
`Demo` function was removed.
Now we offer `sendRandomData` and I do believe it works.

closed 2016-05-13 by alashkin