cesanta/mongoose-os #33: smart.js build for posix platform fails

fedy2, 2015-10-01:
I've tried to build the smart.js for posix platform and ARM architecture but it fails:
```
fedy2@guendalinaXII ~/esp/smart.js/platforms/posix $ sh makearm.sh 
Unable to find image 'docker.cesanta.com:5000/eabihf-arm-build:latest' locally
latest: Pulling from eabihf-arm-build

eeb365049982: Pull complete 
63dd1fb7de4f: Pull complete 
22eedc530341: Pull complete 
95757659f2bd: Pull complete 
de14382e9617: Pull complete 
d12b786894eb: Pull complete 
5c2f1b97b1a5: Pull complete 
bc07cbc8c626: Pull complete 
e58fd13c669a: Pull complete 
981fea4696c5: Already exists 
6071b4945dcf: Already exists 
5bff21ba5409: Already exists 
e5855facec0b: Already exists 
8251da35e7a7: Already exists 
Digest: sha256:7d5ffc336b7876fb552c825970f96faebfb5c7133252006ae0ab110482623a82
Status: Downloaded newer image for docker.cesanta.com:5000/eabihf-arm-build:latest
//bin/bash: line 0: cd: /cesanta/smartjs/platforms/posix: No such file or directory
```

mmikulicic, 2015-10-01:
are you on windows?

fedy2, 2015-10-01:
Nope, Linux Mint 17.1

mmikulicic, 2015-10-01:
Please run:

```
$ docker images | grep eabihf-arm-build
```

fedy2, 2015-10-01:
```
fedy2@guendalinaXII ~/esp/smart.js/platforms/posix $  docker images | grep eabihf-arm-build
docker.cesanta.com:5000/eabihf-arm-build     latest              981fea4696c5        7 weeks ago         507.7 MB
```

mmikulicic, 2015-10-01:
yeah, our script it bad and we should feel bad

closed 2015-10-01 by gelraen