cesanta/mongoose-os #34: Build instruction for posix platform on i386 and amd64 architectures

fedy2, 2015-10-01:
Would be nice to have build instruction or better a script file for the building of smart.js for the posix platform under i386 and amd64 architecture.

mmikulicic, 2015-10-01:
`make` should just work. The reason it doesn't is the same as #33.
If `make` would just work out of the box, would you feel it still needs more documentation?

fedy2, 2015-10-01:
I've tried `make` but I get this:
```
fedy2@guendalinaXII ~/esp/smart.js/platforms/posix $ make
Makefile:31: ../../../v7/scripts/platform.mk: No such file or directory
make: *** No rule to make target `../../../v7/scripts/platform.mk'.  Stop.
```

mmikulicic, 2015-10-01:
please pull and retry

fedy2, 2015-10-01:
Perfect. Is working now.

mmikulicic, 2015-10-01:
Thanks for reporting!

closed 2015-10-01 by mmikulicic

DongDongig, 2016-03-13:
I'm also trying to build smart.js for RPI2....
but i cannot figure out how to build...

Could you guide me to build specifically?

I did install "docker" and execute the "armbuild.sh" but I could see

root@ubuntu:/home/ubuntu/smart.js-beta2/smartjs/platforms/posix# ./makearm.sh   GEN build/build_info.c
Traceback (most recent call last):
  File "../../../common/tools/fw_meta.py", line 192, in <module>
    cmd_gen_build_info(args)
  File "../../../common/tools/fw_meta.py", line 66, in cmd_gen_build_info
    repo = get_git_repo(os.getcwd())
  File "../../../common/tools/fw_meta.py", line 29, in get_git_repo
    raise RuntimeError("%s doesn't look like a Git repo" % path)
RuntimeError: /cesanta/smartjs/platforms/posix doesn't look like a Git repo
make: *** [build/build_info.c] Error 1

And When I put just 
$make
root@ubuntu:/home/ubuntu/smart.js-beta2/smartjs/platforms/posix# make
  CC lnx_gpio.c -> build/lnx_gpio.o
  CC lnx_i2c.c -> build/lnx_i2c.o
  CC lnx_spi.c -> build/lnx_spi.o
  CC main.c -> build/main.o
  CC posix_hal.c -> build/posix_hal.o
In file included from posix_hal.c:33:0:
../../../mongoose/mongoose.h:707:25: fatal error: openssl/ssl.h: No such file or directory
 #include <openssl/ssl.h>
                         ^
compilation terminated.
make: *** [build/posix_hal.o] Error 1

I need to run smart.js on RPI2 or other arm board (32bit)
plz let me know how to do that...
thx in advance!!