cesanta/mongoose-os #39: Websocket support?

x1project, 2015-10-01:
if i flash my esp8266 with smart.js firmware, can i use it like "client can connectable websocket server".
for example 
"\<html>....
\<script type="text/javascript"\>
	var ws = new WebSocket("ws://localhost:8080/websockets");
	ws.onopen = function(something) {
		function something(something);
	}
...."
small html page including above code can be connect ws server.
can i use \<script type="text/javascript"\> .... \<\scprit\> code in my esp8266 flashed with smart.js?

mmikulicic, 2015-10-02:
Hi,

Currently Smart.js has a Websocket client. However it's not hard to implement a websocket server.

It's not clear to me what to do you want exactly. Your code snippet is a html + javascript snippet.

I assume you want to host that HTML on a device running Smart.js, open it on a browser and have your JS code on the user's browser communicate with some code (C or JS?) running on the embedded device. Is that correct?

pastukhov, 2015-10-02:
@mmikulicic Yes, this would be great!

x1project, 2015-10-02:
For example;
I have server and it has a capability of websocket communication. I want to use my esp8266 as a client and i should able to program with javascript language. This is my basic idea.
I ask that; i can use my browser as a websocket client (i dont know this term is correct or not) to connect my server. When i connect my server (http:// myserver. com:8080/websocket/default.html) my browser interpret javascript code in source code of default.html . I am asking; a esp8266 flashed with smartjs, can i use it same javascript code for connection to my server as a client
My general idea's picture is blow.
so maybe my approach is completely wrong way i dont know but i search, any guidance would be wellcome.
[image]

mmikulicic, 2015-10-02:
yes, what you want is a WebSocket client running on your ESP device with Smart.js.

Smart.js does that. You can run JavaScript code on your ESP device. It's practically the same code you'd run on your browser, except the HTML part.

Things like `<script type="text/javascript">` are HTML and do not make sense in this context.
They would make sense only if we were talking about files stored on the ESP device and make it act as a HTTP server; but in that case this is not the JavaScript code that would be running on the device.
Does it make sense?

----

@pastukhov would you mind creating a new issue for tracking your feature request that seems to me unrelated to this ticket ?

x1project, 2015-10-02:
>Smart.js does that. You can run JavaScript code on your ESP device. It's practically the same code you'd run on your browser, except the HTML part.

This sentence is what i want to hear. thanks

mmikulicic, 2015-10-02:
you need to build the firmware with `./make.sh RTOS` to enable the WebSocket API.

This SDK is less stable than the default (OSS).

alashkin, 2016-05-13:
@mmikulicic 
Can we close this issue? Does `<script type="text/javascript">` work in current FW?

mmikulicic, 2016-05-13:
We currently do support Websocket client API, in both C and JS
We support only Websocket server API in C.

The `<script type="text/javascript">` is part of the HTML page that is served by the server (www.myserver.com in the picture), so it has nothing to do with our FW.

Closing

closed 2016-05-13 by mmikulicic