cesanta/mongoose-os #40: Demo function overflow heap when wi-fi not configured in ESP

fedy2, 2015-10-02:
I've run the demo function without configuring the Wifi and I got this error:
```
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
"disp_task"(stack_size = 392,task handle = 3fff58f0) overflow the heap_size.
```
Maybe a check and a "Wifi not configured" error message would be better.

mmikulicic, 2015-10-02:
the demo should not crash even if Wifi is not connected

mmikulicic, 2015-10-02:
some context first: espressif ships 2 SDK: OSS and RTOS.

We recently switched our default build to use the RTOS SDK, but we now switched back because of regressions like this.

Please pull and rebuild your firmware (./make.sh clean required) and try it again.

fedy2, 2015-10-02:
The demo function don't crash immediately. I get this output:
```
smartjs 29152/2280$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
undefined
smartjs 24304/1140$ 
Trap 28: pc=0x4025a380 va=0xf96688
Dumping core to debug output
if you want to dump core, type 'y', or else connect with gdb
```
I've registered the device in the dashboard and I see one value from the device:
```
[{"__name__":"temperature","sensor":"1","src":"//api.cesanta.com/d/dev_36552"},[[1443813318096,25.8976682675]]]
```

alashkin, 2015-10-02:
I tried on my ESP - and cannot reproduce. 
Could you give me some details?

1. What model of ESP do you use?
2. How long it works before crashing?
3. Could you turn debug output on (`Debug.mode(1)`), run again and paste output here?
4. As I understand, you connect Wifi before running demo, yes?

Thank you!

fedy2, 2015-10-02:
1. What model of ESP do you use?

ESP8266 - 01

2. How long it works before crashing?

With last tests immediately.

3. Could you turn debug output on (Debug.mode(1)), run again and paste output here?

I got no output, the console hangs after the demo command outputs:
```
Wifi.status()
got ip
smartjs 29152/2256$ Debug.mode(1)
1
smartjs 29152/2256$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
Pushing metric 
```

4. As I understand, you connect Wifi before running demo, yes?

Yes.

5. Could you paste breakdown output as well?

```
smartjs footprint breakdown
---------------------------------
               .text          .irom0.text    .rodata        .data          .bss           
v7             0              59959          0              0              0              
smartjs        494            52780          0              0              0              
sys            10649          123705         0              0              0              
total          11143          236444         0              0              0              

Firmware size is 396Kb (of 512Kb), 23% is available
```

alashkin, 2015-10-05:
@fedy2,

Everything looks OK
I checked fw on ESP01 - it works for me
And your breakdown is absolutely like mine.

So,
1. Could you upload ./firmware/*.bin here?
2.   btw,
Are you sure, that fw hangs?
```
Wifi.status()
got ip
smartjs 29152/2256$ Debug.mode(1)
1
smartjs 29152/2256$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
Pushing metric 
```
What terminal do you use? Is it Flashnchips?
Could you try to execute any command (`2+2`) - just to check if it is really hangs.

fedy2, 2015-10-05:
I've pulled the new code, compiled and flashed it.
Then I've setup the Wifi and run the demo code, worked fine few times until I've set debug mode to 1 and the console hanged.
I've restarted the device, tried again the demo function but it hangs.

1. http://www.filedropper.com/firmware
2. Yes. I write commands but get nothing back.
3. Yes, I'm using Flashnchips.

alashkin, 2015-10-05:
I've tried you fw. It works :(
_thinking_

```
smartjs 29104/2280$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
undefined
smartjs 24152/1140$ Debug.mode(1)
1
smartjs 27288/1488$ Pushing metric 
Cloud reply: [{"v":1,"src":"//api.cesanta.com/cloud.metrics_db","dst":"//api.cesanta.com/d/dev_84227","resp":[{"id":1430752015187,"status":0}]}]
 
Pushing metric 
Cloud reply: [{"v":1,"src":"//api.cesanta.com/cloud.metrics_db","dst":"//api.cesanta.com/d/dev_84227","resp":[{"id":2060076535589,"status":0}]}]
 
Pushing metric 
Cloud reply: [{"v":1,"src":"//api.cesanta.com/cloud.metrics_db","dst":"//api.cesanta.com/d/dev_84227","resp":[{"id":2035761751436,"status":0}]}]
 
Pushing metric 
Cloud reply: [{"v":1,"src":"//api.cesanta.com/cloud.metrics_db","dst":"//api.cesanta.com/d/dev_84227","resp":[{"id":1298154954620,"status":0}]}]
 
Pushing metric 
Cloud reply: [{"v":1,"src":"//api.cesanta.com/cloud.metrics_db","dst":"//api.cesanta.com/d/dev_84227","resp":[{"id":1684513295838,"status":0}]}]
```

Btw, besides `demo()` does fw work?

fedy2, 2015-10-05:
Yes, I used it to send DHT11 values from the ESP to my site, worked fine with no issues.

I've tried again.
If I run the demo after the flashing operation it works fine and I see the log after set debug mode to 1 (I don't see data in the dashboard chart).
If I reboot the device and run the demo it hangs the device.

alashkin, 2016-04-12:
`demo()` function was removed.
Current version offers `sendRandomData()`, which seems to work

closed 2016-04-12 by alashkin