cesanta/mongoose-os #41: http_eval example fails to compile

fedy2, 2015-10-02:
The http_eval example, https://github.com/cesanta/smart.js/tree/master/platforms/esp8266/examples/http_eval, fails to compile:
```
fedy2@guendalinaXII ~/esp/smart.js/platforms/esp8266 $ sh make.sh 
  CC user/v7_http_eval.c -> build/v7_http_eval.o
user/v7_http_eval.c:2:19: fatal error: osapi.h: No such file or directory
 #include "osapi.h"
                   ^
compilation terminated.
make: *** [build/v7_http_eval.o] Error 1
```

alashkin, 2015-10-02:
Did you try `./make.sh clean && ./make.sh`?

fedy2, 2015-10-02:
Here the output:
```
fedy2@guendalinaXII ~/esp/smart.js/platforms/esp8266 $ ./make.sh clean && ./make.sh
  CC user/c_sscanf.c -> build/c_sscanf.o
  CC user/dht11.c -> build/dht11.o
  CC user/disp_task.c -> build/disp_task.o
  CC user/disp_task_net_test.c -> build/disp_task_net_test.o
  CC user/esp_coredump.c -> build/esp_coredump.o
  CC user/esp_data_gen.c -> build/esp_data_gen.o
  CC user/esp_exc.c -> build/esp_exc.o
  CC user/esp_flash_bytes.c -> build/esp_flash_bytes.o
  CC user/esp_gdb.c -> build/esp_gdb.o
  CC user/esp_gpio.c -> build/esp_gpio.o
  CC user/esp_http_client.c -> build/esp_http_client.o
  CC user/esp_i2c.c -> build/esp_i2c.o
  CC user/esp_sj_hal.c -> build/esp_sj_hal.o
  CC user/esp_sj_wifi.c -> build/esp_sj_wifi.o
  CC user/esp_spi.c -> build/esp_spi.o
  CC user/esp_ssl_krypton.c -> build/esp_ssl_krypton.o
  CC user/esp_timers.c -> build/esp_timers.o
  CC user/esp_uart.c -> build/esp_uart.o
  CC user/libc_replacements.c -> build/libc_replacements.o
  CC user/user_main.c -> build/user_main.o
  CC user/util.c -> build/util.o
  CC user/v7_esp.c -> build/v7_esp.o
  CC user/v7_esp_hw.c -> build/v7_esp_hw.o
  CC user/v7_fs.c -> build/v7_fs.o
  CC user/v7_http_eval.c -> build/v7_http_eval.o
user/v7_http_eval.c: In function 'server_eval':
user/v7_http_eval.c:28:3: error: format '%s' expects argument of type 'char *', but argument 5 has type 'void *' [-Werror=format=]
   snprintf(buf, size + 1, "%.*s", (int) size, body);
   ^
user/v7_http_eval.c:28:3: error: format '%s' expects argument of type 'char *', but argument 5 has type 'void *' [-Werror=format=]
user/v7_http_eval.c:29:3: error: passing argument 2 of 'v7_exec' from incompatible pointer type [-Werror]
   err = v7_exec(v7, &v, buf);
   ^
In file included from user/v7_http_eval.c:11:0:
../../src/v7.h:106:13: note: expected 'const char *' but argument is of type 'v7_val_t *'
 enum v7_err v7_exec(struct v7 *, const char *js_code, v7_val_t *result);
             ^
user/v7_http_eval.c:29:3: error: passing argument 3 of 'v7_exec' from incompatible pointer type [-Werror]
   err = v7_exec(v7, &v, buf);
   ^
In file included from user/v7_http_eval.c:11:0:
../../src/v7.h:106:13: note: expected 'v7_val_t *' but argument is of type 'char *'
 enum v7_err v7_exec(struct v7 *, const char *js_code, v7_val_t *result);
             ^
user/v7_http_eval.c:31:3: error: enumeration value 'V7_STACK_OVERFLOW' not handled in switch [-Werror=switch]
   switch (err) {
   ^
user/v7_http_eval.c:31:3: error: enumeration value 'V7_AST_TOO_LARGE' not handled in switch [-Werror=switch]
user/v7_http_eval.c:31:3: error: enumeration value 'V7_INVALID_ARG' not handled in switch [-Werror=switch]
user/v7_http_eval.c:41:3: error: enumeration value 'V7_STACK_OVERFLOW' not handled in switch [-Werror=switch]
   switch (err) {
   ^
user/v7_http_eval.c:41:3: error: enumeration value 'V7_AST_TOO_LARGE' not handled in switch [-Werror=switch]
user/v7_http_eval.c:41:3: error: enumeration value 'V7_INVALID_ARG' not handled in switch [-Werror=switch]
user/v7_http_eval.c:59:3: error: pointer targets in passing argument 2 of 'espconn_sent' differ in signedness [-Werror=pointer-sign]
   espconn_sent(c, resp, strlen(resp));
   ^
In file included from user/v7_http_eval.c:7:0:
/opt/Espressif/ESP8266_SDK/include/espconn.h:290:7: note: expected 'uint8 *' but argument is of type 'char *'
 sint8 espconn_sent(struct espconn *espconn, uint8 *psent, uint16 length);
       ^
user/v7_http_eval.c: In function 'server_serve':
user/v7_http_eval.c:72:3: error: unknown type name 'spiffs_stat'
   spiffs_stat stat;
   ^
user/v7_http_eval.c:73:3: error: unknown type name 'spiffs_file'
   spiffs_file fd = 0;
[... 52 lines cut ...]
     espconn_sent(c, (char *) server_error, sizeof(server_error));
     ^
In file included from user/v7_http_eval.c:7:0:
/opt/Espressif/ESP8266_SDK/include/espconn.h:290:7: note: expected 'uint8 *' but argument is of type 'char *'
 sint8 espconn_sent(struct espconn *espconn, uint8 *psent, uint16 length);
       ^
user/v7_http_eval.c:110:3: error: pointer targets in passing argument 2 of 'espconn_sent' differ in signedness [-Werror=pointer-sign]
   espconn_sent(c, (char *) buf, resp_size);
   ^
In file included from user/v7_http_eval.c:7:0:
/opt/Espressif/ESP8266_SDK/include/espconn.h:290:7: note: expected 'uint8 *' but argument is of type 'char *'
 sint8 espconn_sent(struct espconn *espconn, uint8 *psent, uint16 length);
       ^
user/v7_http_eval.c:114:3: error: implicit declaration of function 'SPIFFS_close' [-Werror=implicit-function-declaration]
   SPIFFS_close(&fs, fd);
   ^
user/v7_http_eval.c:68:15: error: unused variable 'ok_gzip' [-Werror=unused-variable]
   const char *ok_gzip = "HTTP/1.1 200 OK\r\nContent-Encoding: gzip\n\r\n";
               ^
user/v7_http_eval.c:64:7: error: variable 'res' set but not used [-Werror=unused-but-set-variable]
   int res;
       ^
cc1: all warnings being treated as errors
make: *** [build/v7_http_eval.o] Error 1
```

alashkin, 2016-04-13:
Sorry, we discontinued support of that sample.

closed 2016-04-13 by alashkin