cesanta/mongoose-os #44: Wifi disconnect function

fedy2, 2015-10-02:
Would be useful to have a disconnect function in Wifi API.

santiago, 2016-03-21:
what about the MAC address

mmikulicic, 2016-03-21:
@santiago: the mac address is available in the `Sys.ro_vars.mac_address` constant.

mmikulicic, 2016-03-21:
`Wifi.disconnect()` is implemented. Connect-at-boot setting is controlled by `Sys.wifi.sta.enable` conf var (see configuration UI server at http://192.168.4.1).

@alashkin: my understanding is that when the wifi not actively used the actual wifi circuitry is off; i.e. after Wifi.disconnect() (or after booting when the `wifi.sta.enable` conf is false) the wifi is off and not drawing power. Is that correct ?

alashkin, 2016-03-21:
@mmikulicic 
`Wifi.disconect` doesn't change configuration, i.e. it works until reboot.
so,
```
wifi.sta.enable=false
wifi.ap.enable=false
Sys.conf.save();
```
is required to permanently turn off Wifi

closed 2016-03-21 by mmikulicic