cesanta/mongoose-os #53: Are there timers / setTimeout() or access to millis()?

cottsak, 2015-10-13:
More doco somewhere?

mmikulicic, 2015-10-13:
We aim at implementing a reasonable subset of the ecmascript standard.

Thus we have:

- `new Date().getTime()`
- and `setTimeout()`

Unfortunately we didn't yet document which parts of it are implemented and which not, mostly because we are still in alpha and hopefully we could just defer to sources as MDN for reference documentation of the JS standard library. 

We did however document our C<->JS API and any non standard JS API.

Anyway, I guess it would make sense to extract the list of supported functions from our sources somehow.

closed 2015-10-13 by mmikulicic

cottsak, 2015-10-14:
Great news. 

`setInterval()` too then I assume?

mmikulicic, 2015-10-14:
Unfortunately that one is not in yet, but it should be.