cesanta/mongoose-os #54: Are the Cloud helpers making HTTPS calls?

cottsak, 2015-10-13:
related: and do the `Http.get` helpers, for example, support SSL/TLS out of the box? ie, if I supply a "https://host.." address will it know what to do? I only ask because in [other libs](https://github.com/esp8266/Arduino/issues/43) its not automagic.

Also: is cert pinning easy?

mmikulicic, 2015-10-13:
HTTPS is not yet supported, but we are working on it. When it will be, the https:// prefix will be enough to specify SSL.

cottsak, 2015-10-14:
So all of your `Cloud` interaction is not encrypted? The stuff to/from your dashboard?

mmikulicic, 2015-10-14:
Yeah in the alpha version we didn't manage to squeeze the SSL library because of heap space and other SDK issues. It's a ESP8266 specific issue. The POSIX (raspberry pi, Galileo, etc) port has SSL support.

We obviously do indend to address the security of communication in the end product.

cottsak, 2015-10-15:
@mmikulicic Kool. That's good to hear. I'm keen to give this a go but no HTTPS is a deal breaker. I'll need that to get started.

cottsak, 2015-10-16:
@mmikulicic Consider that in IoT land, security/encryption **can not** be a feature! It has to be baked in and robust.

mmikulicic, 2015-10-16:
I couldn't agree more.

I'd even say more, releasing a half baked encryption is worse than releasing none. 

Feel free to preview the other pieces of our platform and the other ports where SSL is available, keeping in mind try that this is literally an alpha release. Your feedback would be appreciated.

cottsak, 2015-10-16:
@mmikulicic Sounds like we're on the same page. I'll see how I go. Thanks mate.

alashkin, 2016-05-13:
SSL support was introduced.
It is described here: https://docs.cesanta.com/mongoose-iot/latest/#/javascript-api/http/

closed 2016-05-13 by alashkin

cottsak, 2016-05-16:
nice