cesanta/mongoose-os #56: better garbage collection for closures?

danyocom, 2015-10-25:
Running the code below on a NodeMCU board (ESP8266) causes the interpreter to crash at around after calling add around 100 times. 
I have tried several different scenarios using closures and they all lead to smart.js crashing. I have not taken a look yet at the code yet to see how closures are handled internally in smart.js

```javascript
var add = (function () {
    var counter = 0;
    return function () {return counter += 1;}
})();

for(i=0;i<=1000;i++)
{
  console.log(add());
}
```

mmikulicic, 2015-10-25:
This regression was introduced with the StackTrace feature. Please disable it

danyocom, 2015-10-25:
I disabled StackTrace by removing the  -DV7_ENABLE__StackTrace=1  from the Makefile. However, my code still does not execute completely. Below is my output, any suggestions?
```
1 
2 
3 
4 
5 
6 
7 
8 
9 
10 
11 
12 
13 
14 
15 
16 
17 
18 
19 
20 
21 
22 
23 
24 
25 
26 
27 
28 
29 
30 
31 
32 
33 
34 
35 
36 
37 
38 
39 
40 
41 
42 
43 
44 
45 
46 
47 
48 
49 
50 
51 
52 
53 
54 
55 
56 
57 
58 
59 
60 
61 
62 
63 
64 
65 
66 
67 
68 
69 
70 
71 
72 
73 
74 
75 
76 
77 
78 
79 
80 
81 
82 
83 
84 
85 
86 
87 
88 
89 
90 
91 
92 
93 
94 
95 
96 
97 
98 
99 
100 
101 
102 
103 
104 
105 
106 
107 
108 
109 
110 
111 
112 
113 
114 
115 
116 
117 
118 
119 
120 
121 
122 
123 
124 
125 
126 
127 
128 
129 
130 
131 
132 
133 
134 
135 
136 
137 
138 
139 
140 
141 
142 
143 
144 
145 
146 
147 
148 
E:M 24
E:M 24
E:M 24
E:M 24

E:M 32
E:M 32

Trap 9: pc=0x40239a7a va=0x1
Dumping core to debug output
if you want to dump core, type 'y', or else connect with gdb
```

mmikulicic, 2015-10-25:
ok thanks for the report, we'll take a look

danyocom, 2015-10-25:
I was having trouble getting closures to work in some code I was writing, so I decided to find a basic closure example to use as test code (posted above).  It turns out that turning off stack tracing did fix the crashes I was seeing in my other code I was writing (just not in the 'closure example' i posted above).

cpq, 2016-12-27:
Closing as obsolete.

closed 2016-12-27 by cpq