cesanta/mongoose-os #59: Code running on ESP8266 executes slower with latest commits.

danyocom, 2015-10-29:
I pulled the latest smart.js today and noticed that my code (GPIO writing in particular, seems to execute slower). Did something change that would affect the execution speed of the js/c code?

danyocom, 2015-10-30:
I tracked the issue down to the `-finstrument-functions` flag in the `MEMORY_FLAGS` section of `platforms/esp8266/Makefile`

mmikulicic, 2015-10-30:
Yeah, we should improve the makefile. The instrumentation is useful to catch stack overflows

alashkin, 2016-05-20:
@mmikulicic 
At the moment, there is possibility to build FW with `STACK_GUARD=false` , which seems disables instrumentation.
It is enough to close this issue?

mmikulicic, 2016-05-20:
this is one major cause of slowdown. not sure though if this addresses the original issue. @danyocom would you like to try it?

cpq, 2017-02-07:
Closing this as rotten, please reopen if needed.

closed 2017-02-07 by cpq