cesanta/mongoose-os #61: MQTT documentation?

roccomuso, 2015-11-02:
Where do i find documentation on MQTT api?

mmikulicic, 2015-11-02:
Sorry, we didn't yet provide a JS interface for the underlying mongoose MQTT protocol. Keeping this issue open until we do.

pastukhov, 2015-11-10:
+2

tyriis, 2016-02-27:
Will this ever be build? For me rhis is the only reason not giving smart.js a try

mmikulicic, 2016-02-27:
Yes, stay tuned

closed 2016-03-04 by alashkin

mmikulicic, 2016-03-04:
MQTT interface exposed to JS (944253002a8d67ec0485d497178d83c27c57382f and 070cc6524330886d25aaf9effc4bd73278549d02)

The documentation is available at: https://www.cesanta.com/developer/smartjs#_mqtt

You can also fetch a prebuilt fw bundle at: https://backend.cesanta.com/devel/smartjs-esp8266-last.zip

tyriis, 2016-03-04:
Nice! Will check this out soon :)

snsekar, 2016-03-05:
Great...i will check this. 

Our MQTT service is ssl enabled. Can i use ssl certificates with this javascript interface?

snsekar, 2016-03-08:
I checked MQTT with above mentioned build. Working fine. 
But we need ssl support also. Mongoose already have ssl support. So i guess soon smart.js will expose js interface for that.

mmikulicic, 2016-03-08:
using `mqtts://` will cause SJS to enable SSL on the underlying Mongoose connection.
@snekar, on which platform are you using smartjs ?

@rojer, what else is necessary to properly setup SSL support on esp8266 ?

snsekar, 2016-03-08:
@mmikulicic We are using smart.js on esp8266

If i use mqtts:// it throws "unsupported protocol"

I checked the code also. Currently only mqtt:// supported https://github.com/cesanta/smart.js/commit/944253002a8d67ec0485d497178d83c27c57382f

  if (mg_vcmp(&scheme, "mqtt") != 0) {
     rcode = v7_throwf(v7, "Error", "unsupported protocol");
     goto clean;
   }

mmikulicic, 2016-03-08:
ok, that part should be easy to fix

snsekar, 2016-03-08:
ok Thanks. Expecting an update on this soon.

mmikulicic, 2016-03-08:
@snsekar done, please take a look.

keep in mind that the device doesn't have enough space to store the full CA bundle and thus it's not easy to just automatically verify every possible server certificate. We have the ability to upload a CA file specific to the server(s) you want to contact (either via https or mqtts or other protocols) but currently we enable this only for the websocket connection to our servers and we didn't yet offer a way to specify a CA file for user defined connections. It's something we obviously want to complete; stay tuned.

snsekar, 2016-03-09:
@mmikulicic Thanks for quick response

We need peer verification to our server only. So we need to use our own CA, cert and private key in mqtt connection. Size is not more than 5kb.

From your code commit  https://github.com/cesanta/smart.js/commit/eda9f86731c785c7e22be1f49e4bd34cdf3db0c4 at line 183 i am seeing
```
mg_set_ssl(nc, NULL, NULL);
```
**Instead of NULL if i pass cert and CA file path will this work? Can i try this?**

As per mongoose documentation it should work i guess https://www.cesanta.com/developer/mongoose#_core_tcp_udp_ssl

I mean this 
```
const char *mg_set_ssl(struct mg_connection *nc, const char *cert,
                       const char *ca_cert);
```
Enable SSL for a given connection. cert is a server certificate file name for a listening connection, or a client certificate file name for an outgoing connection. Certificate files must be in PEM format. Server certificate file must contain a certificate, concatenated with a private key, optionally concatenated with parameters. ca_cert is a CA certificate, or NULL if peer verification is not required. Return: NULL on success, or error message on error.

mmikulicic, 2016-03-09:
yes sure, if building a custom FW is an option for you, you can do it easily as you described.

What we want to do is to refactor this in order for people to easily use the stock SJS FW.

snsekar, 2016-03-09:
ok thanks. I will try that

piyush-arora, 2016-10-06:
@mmikulicic  , I am using mqtt client in esp8266 to subscribe a topic using javascript api and i am using mosquito broker installed on my server , I works perfectly fine for like 5 minutes and then it stops receiving messages where as the other subscriber like in my android device works for the entire time . It seems to have some memory issue . how can i clean the buffer every time through javascript api as I didnot see any . Please help me , I am stuck here

mmikulicic, 2016-10-06:
does the device crash or does it just stop working?

piyush-arora, 2016-10-06:
it does not crash ,it  just stops working . I am using console. log for every 1 sec to display the receiving data but after 5 min or so i could not see any data in console screen .

mmikulicic, 2016-10-06:
could you please paste the code you're using?

piyush-arora, 2016-10-06:
Wifi.changed(
                  function (event)
                  {
                      Wifi.status();
                      if(event ==1)
                      {
                        print("Connected to Wifi and trying to fetch ip ");
                      }
                      else if(event ==2)
                      {
                          print("got the ip ");
                          print (Wifi.ip());
                          print("connecting to MQTT");
                          ConnectMQTT();
 
                      }
                      
                   }
              );

function ConnectMQTT()
{

  var client = MQTT.connect('mqtt://139.59.4.141:1883');

  client.on('connect', 
                        function() 
                        {
                             print("connected to MQTT");
                             client.subscribe('topicName');
                             client.publish('topicName', 'hi I am switch');
                        }
            );

  client.on('message', 
                        function(topic, message) 
                        {
                             console.log("got", message);
                             // based on that control GPIO pins                            
                        }
            );

  

}

piyush-arora, 2016-10-06:
Any Suggestions ?

mmikulicic, 2016-10-06:
Hm, I'd like to figure out whether you're is a JS problem or a Mongoose MQTT proble.

Do you watch the console output on the uart?

Could you please enable verbose logging ?

```js
Sys.conf.debug.stdout_uart=1;
Sys.conf.save();
```

and possibly also:

```js
Sys.conf.debug.level = 4;
Sys.conf.save();
```

(note, this will produce a lot of logs, please capture the console output in a log file with MFT)

---

BTW, an easier way to start some code when the wifi is ready is:

```js
Wifi.ready(ConnectMQTT);

function ConnectMQTT()
{
......
```

piyush-arora, 2016-10-07:
Hi , When I use

Sys.conf.debug.stdout_uart=1;
Sys.conf.save();

my console output doesnot show anything 

And I dont know how to capture log file using MFT , can you explain what is MFT . Thanks

mmikulicic, 2016-10-07:
yeah, sorry it's ```
Sys.conf.debug.stdout_uart=0;
Sys.conf.debug.stderr_uart=0;
```

What tool do you use to flash the fw?

piyush-arora, 2016-10-07:
I use flash n chips tool given by you guys

mmikulicic, 2016-10-07:
ok, MFT is just the new name of the tool. In Flash'n'chip you have settings menu where you can set the console log file.

piyush-arora, 2016-10-07:
I have copied the console output into a text file 
[log.txt](https://github.com/cesanta/mongoose-iot/files/515501/log.txt)

you will see in line number 552 where I start getting the data from mqtt server 
it looks something like this 

got 1475823954606

but after line number 2503 
I donot receive the data 
This happen like after 2-3 mins 

I can assure you that server side is pushing the data 
Can you find what could be the issue ? Thanks for spending your valuable time on it

mmikulicic, 2016-10-07:
Here is a highlight of what happens:

```
mg_mgr_poll          end poll @100942, 4 conns, 2 timers (min 101905), next in 964 ms
mg_lwip_tcp_recv_cb  0x3fff3de4 0x3fff4074 26 0
mg_recv_common       0x3fff3de4 26 26
mg_call              0x3fff3de4 proto ev=3 ev_data=0x3ffffe90 flags=0 rmbl=52 smbl=0
mg_if_recved         0x3fff3de4 0x3fff4074 26
mg_call              0x3fff3de4 after proto flags=0 rmbl=26 smbl=0
^^^ a receive happened on mongoose connection 0x3fff3de4

got 1475824041719
^^^ your callback

mg_lwip_tcp_recv_cb  0x3fff3de4 0x3fff4074 0 0
mg_close_conn        0x3fff3de4 2048
mg_call              0x3fff3de4 proto ev=5 ev_data=0x0 flags=2048 rmbl=26 smbl=0
mg_call              0x3fff3de4 after proto flags=2048 rmbl=26 smbl=0
mg_if_destroy_conn   0x3fff3de4 tcp_close 0x3fff4074
^^^ the 0x3fff4074 got closed after receiving a packet (was it a FIN packet?)

```

I suggest you add the `.on('error', ...)` and `.on('close', ...)` callback handlers.

piyush-arora, 2016-10-07:
Sure , I will work on it . Thanks