cesanta/mongoose-os #62: Http.createServer error

SasaTepic, 2015-11-06:
Hi, when I try to run example:
```
var server = Http.createServer(function(req, res) {
      if (req.uri == '/hello') {
        res.writeHead(200, {'Content-Type': 'text/plain'});
        res.write(req);
        res.end('');
      } else {
        res.serve({ document_root: '/tmp' });
      }
    }).listen('127.0.0.1:8000');
```
or just enter this in console:
```
Http.Createserver();
```
I get folowing error:
```
smartjs 29064/2256$ Http.createServer()
Exec error:{"message":"value is not a function"}
smartjs 29064/2256$ 
```

I'm using latest beta on esp8266 board.

pabbott0, 2015-11-07:
If you do a File.list('*') in the Flash'n'Chips console on the latest esp8266 release for OSX, I'm betting that you'll see that a load of .js files are missing compared to what's in git. Haven't checked the builds for other platforms.

I managed to get a cloud api update accepted by typing out an Http.post() longhand in the console. This was fun because I kept getting stack overflows when the console tried to parse the nested JSON I was feeding it. Breaking up the nesting into multiple vars worked without blowing up the interpreter.

mmikulicic, 2016-04-08:
This is fixed in the latest SJS release, sorry for not updating this issue once fixed.

closed 2016-04-08 by mmikulicic