cesanta/mongoose-os #64: Demo() crashes on an ESP8266

nkolban, 2015-11-14:
I am a new user and am following the instructions trying to get demo() running.  Here is the log of what I entered and, as we can see, it causes the ESP8266 to crash.  If there are other tests I can run, please let me know.

```
Device id: //api.cesanta.com/d/dev_****
Device psk: *****
Cloud: //api.cesanta.com 
HELO! Type some JS. See https://github.com/cesanta/smart.js for more info. 

smartjs 29072/1008$ Joining sweetie 
Wifi.setup('sweetie', 'kolbanpassword')
true
smartjs 28752/1200$ demo
[function demo()]
smartjs 28752/1200$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
fun 0x40237bf4 sp 0x3fffdfc0 limit 0x3fffdfa0 left 32

Trap 9: pc=0x4026ae4e va=0x1
Dumping core to debug output
rebooting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29672, room 16 
tail 8
chksum 0x54
load 0x3ffe8000, len 2480, room 0 
tail 0
chksum 0x70
load 0x3ffe89b0, len 2052, room 8 
tail 12
chksum 0x02
csum 0x02
don't use rtc mem data
```

mmikulicic, 2015-11-18:
@nkolban reproduced, we're looking at it

closed 2016-02-18 by dimonomid

cjdg, 2016-03-14:
Confirmed

dimonomid, 2016-03-14:
@cjdg, could you please elaborate?

Smart.js configuration has changed dramatically since this issue was created; the reason it was crashing is already eliminated a long time ago, and even more, it does not have any `demo()` function anymore, unless user defines his own function, which can do whatever.

What do you mean "Confirmed"?