cesanta/mongoose-os #65: Can't seem to stop previous program from running

nkolban, 2015-11-14:
I created a file on the ESP8266 called 'http_client_1.js' as per the documentation here:

https://www.cesanta.com/developer/smartjs#_http_client_example

I then ran that file with

```
File.eval('http_client_1.js')
```

It seemed to run correctly.  Great!!!

However, it now seems that when ever I restart or reboot my ESP8266, this application seems to automatically launch and run.  How do I stop the application from executing?

cpq, 2015-11-23:
Hi Neil,
great that it works as expected!
Could you clarify on whether you put `File.eval('http_client_1.js')` call in one of the initialization files, please?

nkolban, 2015-11-23:
I am pretty sure that I merely entered it through the serial terminal.   However, I'll try again tonight ... what I'll do is 100% erase my flash and follow again the steps that got me to where I was.

cpq, 2015-11-23:
Ok, thank you Neil. Please follow up as soon as you manage to reproduce.

nkolban, 2015-11-23:
Howdy ... I have managed to recreate.  Here is the exact story.
The version of Smart.js being used is:

    ESP8266-p/20151112-135329/f829b921

I have a file called `hc1.js` that contains:

```
function report() {
  print('sending request...');
  Http.request({
    hostname: 'httpbin.org',
    path: '/get?n=' + GC.stat().sysfree  // Free RAM
  }, function(response) { print(response); }).end(/* no POST data */);

  setTimeout(report, 3000);  // Report every 3 seconds
};

report();
```

I use Flash'n'Chips version `0.1` to upload the file to Smart.js (running on ESP8266).

I check that the file is present using:

    File.list('.')

I run the script using:

    File.eval('hc1.js')

All runs and it looks good.

I reboot my device ...

Here is what I see on my console:

```
�d�8��og��$`don't use rtc mem data
;l�cannot save conf 
 
Device id: //api.cesanta.com/d/dev_XXX
Device psk: psk_XXX
Cloud: //api.cesanta.com 
sending request... 

smartjs 27424/504$ sending request... 
sending request... 
sending request... 
sending request... 
sending request

```

Let me know what other info might be needed.  At y'alls disposal.

nkolban, 2015-11-24:
I have built the latest Smart.js for the ESP8266 from source and flashed my ESP8266 ... I am now no longer able to reproduce the error.  As such, I am going to assume that between the build I had previously installed and the current build:

    "ESP8266-master/20151124-014710/7e198c38"

Something got cured.   Either way, I am happy now ... closing issue.

closed 2015-11-24 by nkolban