cesanta/mongoose-os #67: Flashnschips fails flashing ESP8266 03 firmware

fedy2, 2015-11-15:
I'm trying to flash my ESP8266 03 device.
I run the flashnchips tool with this command:

    ./flashnchips --gui --console-baud-rate 76800  --flash-baud-rate 76800 -d

If I connect the console and reboot the device I get this line:

    ets Jan  8 2013,rst cause:1, boot mode:(1,0)

If I try to flash the firmware I get a "failed to talk to the boot loader" error. In the console I get this logs:
`
DEBUG: Adding ports: QSet("ttyUSB0")
INFO: MainDialog state changed from 0 to 1
INFO: MainDialog state changed from 1 to 0
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Writing bytes "ttyUSB0" : "000a0400000000005000f03f"
DEBUG: Read bytes "ttyUSB0" : "010a02000000569c0000"
DEBUG: Writing bytes "ttyUSB0" : "000a0400000000005400f03f"
DEBUG: Read bytes "ttyUSB0" : "010a020046fe00020000"
INFO: MAC address:  "18fe34fe469c"
INFO: MainDialog state changed from 0 to 1
INFO: Loading "0x00000.bin"
INFO: Loading "0x11000.bin"
INFO: Loading "0x70000.bin"
INFO: MainDialog state changed from 1 to 2
INFO: MainDialog state changed from 2 to 3
WARNING: Baud rate of serial port /dev/ttyUSB0 is set to 76923 instead of 76800: divisor 312.500000 unsupported
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
INFO: MainDialog state changed from 3 to 2
INFO: MainDialog state changed from 2 to 1
`

mmikulicic, 2015-11-15:
Hi,

Does it also fail with default baud rates?

fedy2, 2015-11-15:
Yes. Here the log:
```
fedy2@guendalinaXII ~/esp/smart.js/flashnchips $ ./flashnchips --gui -d 
DEBUG: Adding ports: QSet("ttyUSB0")
INFO: MainDialog state changed from 0 to 1
INFO: MainDialog state changed from 1 to 0
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Writing bytes "ttyUSB0" : "000a0400000000005000f03f"
DEBUG: Read bytes "ttyUSB0" : "010a02000000569c0000"
DEBUG: Writing bytes "ttyUSB0" : "000a0400000000005400f03f"
DEBUG: Read bytes "ttyUSB0" : "010a020046fe00020000"
INFO: MAC address:  "18fe34fe469c"
INFO: MainDialog state changed from 0 to 1
INFO: Loading "0x00000.bin"
INFO: Loading "0x11000.bin"
INFO: Loading "0x70000.bin"
INFO: MainDialog state changed from 1 to 2
INFO: MainDialog state changed from 2 to 3
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
INFO: MainDialog state changed from 3 to 2
INFO: MainDialog state changed from 2 to 1
```

gelraen, 2015-11-16:
@fedy2 please try to reboot the device manually between pressing "Detect devices" and "Flash firmware". Also, since you have only one serial port, you don't really need to use "Detect devices", it just tells you on which ports the devices are detected.

gelraen, 2015-11-16:
For some reason `trySync` function fails when trying to flash after "Detect" was pressed without rebooting the device afterwards. I don't get yet what's going on, pressing "Detect" without rebooting works and payload in both cases seems to be identical.

gelraen, 2015-11-16:
Aha, "Detect devices" opens the port at 9600 each time, while "Flash firmware" tries to share the same handler with console at different speed. It seems that ESP detects baud rate only once after boot, so "Detect devices" makes it believe that we will use 9600 all the time and any further attempts to issue sync command at 230400 fail.

Not yet sure what to do about it, but F&C should handle this case better.

fedy2, 2015-11-16:
Hi, I'm using the "Detect devices" function only to check if the application can communicate with the device (A check symbol appears after the check on the select box option, otherwise nothing happens). What you mean with "manual reboot"? I'm simply powering the device, not the adapter, off and on again.

gelraen, 2015-11-16:
By "manual reboot" I mean just power cycle ESP (e.g. disconnect and connect back VCC pin) after you get that check mark and before pressing "Flash firmware"

fedy2, 2015-11-16:
With "manual reboot" I get the same error:
```
fedy2@guendalinaXII ~/esp/smart.js/flashnchips $ ./flashnchips --gui -d 
DEBUG: Adding ports: QSet("ttyUSB0")
INFO: MainDialog state changed from 0 to 1
INFO: MainDialog state changed from 1 to 0
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Read bytes "ttyUSB0" : "01080200070712200000"
DEBUG: Writing bytes "ttyUSB0" : "000a0400000000005000f03f"
DEBUG: Read bytes "ttyUSB0" : "010a02000000569c0000"
DEBUG: Writing bytes "ttyUSB0" : "000a0400000000005400f03f"
DEBUG: Read bytes "ttyUSB0" : "010a020046fe00020000"
INFO: MAC address:  "18fe34fe469c"
INFO: MainDialog state changed from 0 to 1
INFO: Loading "0x00000.bin"
INFO: Loading "0x11000.bin"
INFO: Loading "0x70000.bin"
INFO: MainDialog state changed from 1 to 2
INFO: MainDialog state changed from 2 to 3
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
DEBUG: Writing bytes "ttyUSB0" : "0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555"
DEBUG: No data:  "Unknown error"
DEBUG: Incomplete response:  ""
INFO: MainDialog state changed from 3 to 2
INFO: MainDialog state changed from 2 to 1
```

gelraen, 2015-11-16:
Just to be clear, you did reboot your ESP after this line: `INFO: MAC address:  "18fe34fe469c"` and it still didn't work? I'll try to check if I can reproduce it tomorrow.

fedy2, 2015-11-16:
Yes, I've rebooted the device after the INFO: MAC line.

gelraen, 2015-11-17:
Apparently, leaving GPIO2 connected to VCC somehow makes ESP still remember the baud rate. Please try to either disconnect all pins connected to VCC or grounding RST pin for a moment. Each of those things makes it flash properly after detection for me.

fedy2, 2015-11-19:
I've not found the RST pin on the ESP 03 device, but disconnecting the CH_PD pin for a moment from the VCC worked fine. Please consider updating the documentation for the ESP device.

Thank you very much for the support!

gelraen, 2015-11-19:
You're welcome :) And thanks for testing and reporting problems!

New Flashnchips will simply reboot ESP after probing, which should workaround this issue.

closed 2015-11-19 by gelraen