cesanta/mongoose-os #68: error building flashnchips on Ubuntu

eallik, 2015-11-19:
I have Qt5, `qt5serialport5-dev` as well as `qt5-make` installed

```
$ uname -a
Linux PW-cmp 3.13.0-39-generic #66-Ubuntu SMP Tue Oct 28 13:30:27 UTC 2014 x86_64 x86_64 x86_64 GNU/Linux
$ qmake -v
QMake version 3.0
Using Qt version 5.2.1 in /usr/lib/x86_64-linux-gnu
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.1 LTS
Release:	14.04
Codename:	trusty
```

and here's the error I get:

```
admin@PW-cmp:~/code/smart.js/flashnchips$ qmake
admin@PW-cmp:~/code/smart.js/flashnchips$ make
/usr/lib/x86_64-linux-gnu/qt5/bin/uic main.ui -o ui_main.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic about.ui -o ui_about.h
g++ -c -m64 -pipe -O2 -Wall -W -D_REENTRANT -fPIE -DVERSION=\"0.1\" -DAPP_NAME=\"flashnchips\" -DSPIFFS_TEST_VISUALISATION=1 -DSPIFFS_HAL_CALLBACK_EXTRA=1 -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -I. -I. -Icommon/.. -I../src/spiffs -I../tools -I/usr/include -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o cli.o cli.cc
cli.cc:123:3: warning: identifier ‘nullptr’ is a keyword in C++11 [-Wc++0x-compat]
   if (hal_ == nullptr) {
   ^
In file included from ./common/util/status.h:21:0,
                 from cli.h:9,
                 from cli.cc:1:
./common/util/logging.h:55:41: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11 [enabled by default]
   LogMessage(const LogMessage& other) = delete;
                                         ^
./common/util/logging.h:61:21: warning: override controls (override/final) only available with -std=c++11 or -std=gnu++11 [enabled by default]
   ~LogMessageAndDie() override;
                     ^
In file included from hal.h:9:0,
                 from cli.h:11,
                 from cli.cc:1:
prompter.h:15:65: error: ‘>>’ should be ‘> >’ within a nested template argument list
       QString text, QList<QPair<QString, QMessageBox::ButtonRole>> buttons) = 0;
                                                                 ^
prompter.h: In constructor ‘Prompter::Prompter(QObject*)’:
prompter.h:11:67: error: ‘>>’ should be ‘> >’ within a nested template argument list
     qRegisterMetaType<QList<QPair<QString, QMessageBox::ButtonRole>>>(
                                                                   ^
In file included from cli.h:11:0,
                 from cli.cc:1:
hal.h: At global scope:
hal.h:18:11: error: ‘unique_ptr’ in namespace ‘std’ does not name a type
   virtual std::unique_ptr<Flasher> flasher(Prompter *prompter) const = 0;
           ^
In file included from cli.cc:1:0:
cli.h:30:3: error: ‘unique_ptr’ in namespace ‘std’ does not name a type
   std::unique_ptr<HAL> hal_;
   ^
In file included from cli.cc:15:0:
cc3200.h:12:1: error: ‘unique_ptr’ in namespace ‘std’ does not name a type
 std::unique_ptr<HAL> HAL();
 ^
In file included from esp8266.h:9:0,
                 from cli.cc:16:
./common/util/statusor.h:36:20: error: expected ‘,’ or ‘...’ before ‘&&’ token
   inline StatusOr(T&& value);
                    ^
./common/util/statusor.h:39:44: warning: defaulted and deleted functions only available with -std=c++11 or -std=gnu++11 [enabled by default]
   inline StatusOr(const StatusOr& other) = default;
                                            ^
./common/util/statusor.h:40:27: error: expected ‘,’ or ‘...’ before ‘&&’ token
   inline StatusOr(StatusOr&& other);
[... 245 lines cut ...]
   std::unique_ptr<Flasher> f(hal_->flasher(prompter_));
                          ^
cli.cc:173:30: error: ‘hal_’ was not declared in this scope
   std::unique_ptr<Flasher> f(hal_->flasher(prompter_));
                              ^
cli.cc:173:54: error: ‘f’ was not declared in this scope
   std::unique_ptr<Flasher> f(hal_->flasher(prompter_));
                                                      ^
cli.cc:187:3: error: ‘unique_ptr’ is not a member of ‘std’
   std::unique_ptr<QSerialPort> s(r.ValueOrDie());
   ^
cli.cc:187:30: error: expected primary-expression before ‘>’ token
   std::unique_ptr<QSerialPort> s(r.ValueOrDie());
                              ^
cli.cc:187:48: error: ‘s’ was not declared in this scope
   std::unique_ptr<QSerialPort> s(r.ValueOrDie());
                                                ^
cli.cc:202:3: warning: lambda expressions only available with -std=c++11 or -std=gnu++11 [enabled by default]
   });
   ^
cli.cc:213:3: warning: lambda expressions only available with -std=c++11 or -std=gnu++11 [enabled by default]
   });
   ^
make: *** [cli.o] Error 1
```

gelraen, 2015-11-19:
Looks like c++14 is only supported by Qt 5.4 and higher. Can you try replacing `CONFIG += c++14` with `CONFIG += c++11` in `flashnchips.pro`?

Initially it was written with Qt 5.4, so more incompatibilities may come up even with proper `-std` flag.

eallik, 2015-11-19:
thanks! a new error now:

(I also installed and enabled GCC 4.9)

```
admin@PW-cmp:~/code/smart.js/flashnchips$ make distclean
rm -f qrc_blobs.cpp qrc_images.cpp
rm -f moc_cli.cpp moc_dialog.cpp moc_flasher.cpp moc_prompter.cpp moc_sigsource.cpp
rm -f dialog.moc esp8266.moc cc3200.moc sigsource_unix.moc
rm -f ui_main.h ui_about.h
rm -f cli.o dialog.o esp8266.o flasher.o fs.o main.o serial.o cc3200.o error_codes.o logging.o status.o spiffs_cache.o spiffs_gc.o spiffs_nucleus.o spiffs_check.o spiffs_hydrogen.o sigsource_unix.o qrc_blobs.o qrc_images.o moc_cli.o moc_dialog.o moc_flasher.o moc_prompter.o moc_sigsource.o
rm -f *~ core *.core
rm -f flashnchips 
rm -f Makefile
admin@PW-cmp:~/code/smart.js/flashnchips$ qmake
admin@PW-cmp:~/code/smart.js/flashnchips$ make
/usr/lib/x86_64-linux-gnu/qt5/bin/uic main.ui -o ui_main.h
/usr/lib/x86_64-linux-gnu/qt5/bin/uic about.ui -o ui_about.h
g++ -c -m64 -pipe -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIE -DVERSION=\"0.1\" -DAPP_NAME=\"flashnchips\" -DSPIFFS_TEST_VISUALISATION=1 -DSPIFFS_HAL_CALLBACK_EXTRA=1 -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_NETWORK_LIB -DQT_SERIALPORT_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++-64 -I. -I. -Icommon/.. -I../src/spiffs -I../tools -I/usr/include -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtNetwork -I/usr/include/qt5/QtSerialPort -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o cli.o cli.cc
cli.cc: In constructor ‘CLI::CLI(QCommandLineParser*, QObject*)’:
cli.cc:41:40: error: no matching function for call to ‘QTimer::singleShot(int, CLI*, void (CLI::*)())’
   QTimer::singleShot(0, this, &CLI::run);
                                        ^
cli.cc:41:40: note: candidates are:
In file included from /usr/include/qt5/QtCore/QTimer:1:0,
                 from cli.cc:9:
/usr/include/qt5/QtCore/qtimer.h:81:17: note: static void QTimer::singleShot(int, const QObject*, const char*)
     static void singleShot(int msec, const QObject *receiver, const char *member);
                 ^
/usr/include/qt5/QtCore/qtimer.h:81:17: note:   no known conversion for argument 3 from ‘void (CLI::*)()’ to ‘const char*’
/usr/include/qt5/QtCore/qtimer.h:82:17: note: static void QTimer::singleShot(int, Qt::TimerType, const QObject*, const char*)
     static void singleShot(int msec, Qt::TimerType timerType, const QObject *receiver, const char *member);
                 ^
/usr/include/qt5/QtCore/qtimer.h:82:17: note:   candidate expects 4 arguments, 3 provided
make: *** [cli.o] Error 1
```

gelraen, 2015-11-19:
Meh, `QTimer::singleShot` overloads that accept function pointers were added in Qt 5.4. Looks like I need to actually go and backport Flashnchips to Qt 5.2 :)

eallik, 2015-11-19:
Is that a longer process in this case? i.e. should I just hold on for 5-15 min or just forget about it until it's sorted?

gelraen, 2015-11-19:
It will take a day or two I think. After `QTimer::singleShot` I've found that `QCommandLineParser::addOptions` also appeared after Qt 5.2, and that would be easier to fix after another piece of refactoring I'm doing right now.

eallik, 2015-11-19:
I see! Meanwhile, Marko and I are trying to figure out how to best get Qt 5.4 installed on 14.04 LTS — is the official binary installer download the only option? Or do any members of your team know an `apt-get` based path?

gelraen, 2015-11-19:
I have no idea, sorry. On Mac OS X I'm using the binary installer, which neatly puts multiple versions side by side in my home directory.

gelraen, 2015-11-19:
Quick search shows this: https://launchpad.net/~hedges/+archive/ubuntu/qt5.5

mmikulicic, 2015-11-19:
Workaround:

You need to install they Qt5.5 backport available at https://launchpad.net/~hedges/+archive/ubuntu/qt5.5

```
$ apt-get install qt55declarative qt55serialport libgl1-mesa-dev libftdi-dev
$ export PATH=/opt/qt55/bin:$PATH
```

eallik, 2015-11-19:
OK, I have it running now with 5.5; only need to find out why the NodeMCU device is not being detected. :)

mmikulicic, 2015-11-23:
Since 32fc09a07268283f0ca99cf982502038be725d69 FlashNChips can build with Qt 5.2.

mmikulicic, 2015-11-23:
@eallik if you still have any problems with your NodeMCU not being detected, would you mind opening another issue ?

closed 2015-11-23 by mmikulicic

eallik, 2015-11-23:
Thank you very much! What a reaction time!