cesanta/mongoose-os #70: Sometimes File upload in FlashNChips fails and requires to re-flash the device

fedy2, 2015-11-20:
I'm using the file upload feature from Flashnchips. Sometimes it fails hanging during the upload and I need to restart the ESP device to get control back.
When the device is restarted the Fnc tool try to continue the upload, failing because the uf object is not more defined.

If I try again to upload the file I get an error message for each write it try to accomplish:

```
smartjs 29560/1296$ uf.write('\x28\x29\x3b\x0a\x20\x20\x7d\x0a\x7d\x29\x2e\x6c\x69\x73\x74\x65\x6e\x28\x38\x31\x29\x3b')
Exec error: {"message":"value is not a function"}
smartjs 29560/1296$ uf.close()
Exec error: {"message":"value is not a function"}
```

I've restarted both the device and the fnc tool various time but with no success.

Finally I've flashed the device again and the file upload feature of fnc returned to work. 
Note: during the flashing operation I got the following error in logs with relative popup message: 
```
CRITICAL: Failed to read sys_config.json
```

alirezza, 2015-11-21:
I have exactly the same problem,
I'm not able to delete the file with File.remove("") any more.

mmikulicic, 2015-11-21:
sounds like spiffs corruption. Could you dump the content of the FS image with `esptool.py read_flash` and share it with us ?

The address of the FS depends on your build version, look in your `firmware` dir, it's the higher hex number

Coaxial, 2016-02-17:
I ran into the issue, here is a dump:
[smartjs_dump.bin.zip](https://github.com/cesanta/smart.js/files/133645/smartjs_dump.bin.zip)

I hope I did it right, I dumped 512,000 bytes from address `0x73000` which is the highest address in the firmware's directory.

alashkin, 2016-05-20:
FNC and FW receives significant changes since that time.
I'm closing issue, but feel free to reopen, if it is still actual.

closed 2016-05-20 by alashkin