cesanta/mongoose-os #76: Unable to run http_client_1.js

nkolban, 2015-11-23:
I have uploaded the sample called `http_client_1.js` but I get the following error when I try and run it:

```
Exec error:{"message":"value is not a function"}
   at: [function report()]
```

Here is a complete log of what I did and the results.  If there is more I can supply, please don't hesitate to contact me:

```
smartjs 29000/2256$ var f = File.open('http_client_1.js');
{}
smartjs 29000/2196$ f.readAll()
function report() {
  print('sending request...');
  Http.request({
    hostname: 'httpbin.org',
    path: '/get?n=' + GC.stat().sysfree  // Free RAM
  }, function(response) { print(response); }).end(/* no POST data */);

  setTimeout(report, 3000);  // Report every 3 seconds
};

report();
smartjs 28968/2196$ File.eval('http_client_1.js')
sending request... 
Exec error:{"message":"value is not a function"}
   at: [function report()]
smartjs 28472/1800$ 
```

alirezza, 2015-11-23:
I don't know why, But if you reset or reflash the device, It will work.

mmikulicic, 2015-11-23:
It reminds me of a problem with our mmap implementation that failed when spiffs triggered a GC on the flash.

We fixed it in db58c92f5aba16a1f5f23a21986953cd5c6af274

The beta release was before that fix :-(

Please try HEAD or download the developer snapshots (links available at https://github.com/cesanta/smart.js)

mmikulicic, 2015-11-23:
Some context:

Some system functionality is implemented in JS. In order to save ram we want to execute the code right off the flash. But we didn't want to tie our V7 interpreter on specifics of a single port. The V7 interpreter has support for memory mapping a precompiled file and executing from that memory area.

We implemented a simple shim for mmap for esp8266, but in order to make it perform reasonably we had to bypass the spiffs layer and read straight from flash. However spiffs can relocate the filesystem blocks potentially at every write, so we had to hook into spiffs and fixup the mapping between file content and flash location when it changes.

nkolban, 2015-11-23:
I have switched over to `ESP8266-p/20151112-135329/f829b921` which appears to be the latest pre-built firmware.  That is now exhibiting a different symptom (see issue #65).   I feel that running the later version has indeed got me past the issue reported here but let me get to end of job on #65 before closing this one out.

A thought though ... if running the build known as "Smart.js Beta" as found here:

https://github.com/cesanta/smart.js/releases

is known to contain some severity 1 defects, y'all might want to refresh that beta with a new build ... or keep an issues log of known severity 1 defects and workarounds so that virgin users like myself aren't left hanging.

nkolban, 2015-11-24:
After switching to the latest source build represented as:

    "ESP8266-master/20151124-014710/7e198c38"

I am no longer able to reproduce the issue.  As such I am pleased and will close the issue.

closed 2015-11-24 by nkolban