cesanta/mongoose-os #78: "No rule to make target ..." building ESP8266 fw on OSX

eallik, 2015-11-23:
I have Docker installed via Homebrew Cask.

```
~/code/esp8266/smartjs/smart.js/platforms/esp8266$ make
Unable to find image 'docker.cesanta.com:5000/esp8266-build-oss:1.4.1-cesanta-r2' locally
1.4.1-cesanta-r2: Pulling from esp8266-build-oss

0a85502c06c9: Pull complete 
0998bf8fb9e9: Pull complete 
a6785352b25c: Pull complete 
e9ae3c220b23: Pull complete 
65b44a851f4f: Pull complete 
8c3a50b946e3: Pull complete 
050f18e8065a: Pull complete 
811a79941ca0: Pull complete 
fafac5c336db: Pull complete 
c0962a716d2e: Pull complete 
20c245076d3e: Pull complete 
2b485eda8667: Pull complete 
39643645b81f: Pull complete 
0f7be26676d4: Pull complete 
01e69121302f: Pull complete 
ac92877d1e8f: Pull complete 
34d7e1583b30: Pull complete 
45d265b0a128: Pull complete 
b4dcc5b88108: Pull complete 
1a38c390d915: Pull complete 
Digest: sha256:d64d907078130194f48aced399564af94a8ea0b9f947fa3fdbf47b2fd811db3f
Status: Downloaded newer image for docker.cesanta.com:5000/esp8266-build-oss:1.4.1-cesanta-r2
Makefile.build:10: ../../../common/platforms/esp8266/common.mk: No such file or directory
make: *** No rule to make target `../../../common/platforms/esp8266/common.mk'.  Stop.
make: *** [all] Error 2
```

eallik, 2015-11-23:
I'm not sure I understand this line in `Makefile.build`:

```
include ../../../common/platforms/esp8266/common.mk
```

— the `../../..` bit is already pointing above the `smart.js` repo clone.

Also in `Makefile`, there are some pointers to `/cesanta` — is that by design?

nkolban, 2015-11-23:
I just tried on Ubunto and I got the same:

```
kolban@kolban-VirtualBox:~/git/smart.js/platforms/esp8266$ sudo make
Unable to find image 'docker.cesanta.com:5000/esp8266-build-oss:1.4.1-cesanta-r2' locally
1.4.1-cesanta-r2: Pulling from esp8266-build-oss
0a85502c06c9: Pull complete 
0998bf8fb9e9: Pull complete 
a6785352b25c: Pull complete 
e9ae3c220b23: Pull complete 
65b44a851f4f: Pull complete 
8c3a50b946e3: Pull complete 
050f18e8065a: Pull complete 
811a79941ca0: Pull complete 
fafac5c336db: Pull complete 
c0962a716d2e: Pull complete 
20c245076d3e: Pull complete 
2b485eda8667: Pull complete 
39643645b81f: Pull complete 
0f7be26676d4: Pull complete 
01e69121302f: Pull complete 
ac92877d1e8f: Pull complete 
34d7e1583b30: Pull complete 
45d265b0a128: Pull complete 
b4dcc5b88108: Pull complete 
1a38c390d915: Pull complete 
Digest: sha256:d64d907078130194f48aced399564af94a8ea0b9f947fa3fdbf47b2fd811db3f
Status: Downloaded newer image for docker.cesanta.com:5000/esp8266-build-oss:1.4.1-cesanta-r2
Makefile.build:10: ../../../common/platforms/esp8266/common.mk: No such file or directory
make: *** No rule to make target `../../../common/platforms/esp8266/common.mk'.  Stop.
Makefile:9: recipe for target 'all' failed
make: *** [all] Error 2
kolban@kolban-VirtualBox:~/git/smart.js/platforms/esp8266$ 
```

mmikulicic, 2015-11-24:
fixed

closed 2015-11-24 by mmikulicic

nkolban, 2015-11-24:
From the perspective of a user who had this problem before it was fixed ... what should he do to take advantage of the fix?

mmikulicic, 2015-11-24:
Hi Neil,

The user should just pull the most recent master