cesanta/mongoose-os #79: Suggestion: Constant definitions for function calls

nkolban, 2015-11-24:
In a number of places we can supply numeric encodings for function parameters.  For example in GPIO.setmode(), the mode can be 0, 1, 2 or 3.   This issue requests the evaluation of constant definitions for these values.   They may be strings ... or (my preference) object constants.

Example ...

    GPIO.setmode(4, GPIO.modeOutput, GPIO.pullUp);

I am sure that there will be other places throughout the Smart.js APIs where constants could be defined so this Issue can be updated with other notions as needed.

mmikulicic, 2015-11-24:
since we finally can keep strings on flash, now it's possible to avoid wasting ram for the hundreds of bytes needed for all the constants and use them again.

mmikulicic, 2016-02-18:
Added `Debug.OFF`, `Debug.OUT`, `Debug.ERR`

@dimonomid, would you please make sure that we also add some constants to `GPIO` ?

alashkin, 2016-04-13:
Fixed, see description here: https://github.com/cesanta/smart.js/blob/master/docs/javascript-api/gpio.md

closed 2016-04-13 by alashkin