cesanta/mongoose-os #81: Request: Websockets support

nkolban, 2015-11-27:
I'm thinking that Smart.js's web server is likely based on Mongoose and Mongoose has WebSockets support.  If this is the case, would it be possible to expose Websocket support through JavaScript.

mmikulicic, 2015-11-27:
Hi @nkolban. The WebSocket client is already exported. It uses the standard WebSocket API (see https://developer.mozilla.org/en-US/docs/Web/API/WebSocket).

Note to self: document it!

We currently have no WebSocket server support; we just didn't figure out yet what is the best API, but it shouldn't be hard to implement. For other APIs we could easily pick a standard or at least the de facto standard. For WebSocket servers there seems to be many competing APIs.

nkolban, 2015-11-28:
Very nice.  Looking forward to reading about WebSocket client request from Smart.js ... however I wasn't specific in my original post.  The key function I think I am looking for is WebSocket server support.   Let me give you a trivial example.  And it is just that ... an example.

I wish to create an ESP8266 robot chassis.  I want to send commands to the robot to turn or move forwards or backwards.  I wish to use a web page with UI widgets on it and interaction with the widgets (by the end user) will send messages to the ESP8266 to perform the commands.  While I fully realize that each message could be sent as a separate REST request to an HTTP server running under Smart.js ... my "thinking" is that I would like to use WebSocket client calls FROM my browser hosted application to Smart.js which is hosting a WebSocket server.   It appears that full Mongoose can do this ... see `mg_set_protocol_websocket()` which appears to expose a number of `MG_EV_WEBSOCKET_*` events.  The desire here would be that these would be exposed upwards to Smart.js.

mmikulicic, 2016-02-18:
for the record, WebSocket Client API is now documented, changing the title to be more explicit about WebSocket server.

carterw, 2016-05-30:
It doesn't make sense to me to have the 8266 act as a websocket server. You can implement the websocket server on some other machine and attach to that using the existing 8266 websocket client API. Then you could have multiple robots talking to your server.

mmikulicic, 2016-05-31:
Yeah, devices will often be behind NAT anyway, and also they might be offline (e.g. sleeping). It generally makes more sense for the devices to be clients and connect to some server or some gateway.

That said, Mongoose IoT doesn't only support ESP8266. Other than the small embedded platforms, Mongoose IoT can work on any posix system. You can also run the same software stack on one of such gateways (e.g. a router, a RPI, etc). Thus, the feature request is reasonable, but it's currently low priority for us.

cpq, 2017-03-02:
https://mongoose-os.com/video-tutorials.html#video6

Although it's not a low level websocket api, the RPC mechanism implements very practical way of controlling the device. No need to invent the wheel and care about frame format, commands queue, confirmations, etc.

cpq, 2017-04-17:
Closing for now, as there are no short or mid term plans w.r.t. WS server API.

closed 2017-04-17 by cpq