cesanta/mongoose-os #82: Request: Support for NeoPixels

nkolban, 2015-11-28:
This is a strange request ... I have been working hard on other environments on ESP8266 and what I kept finding was that my user bases kept asking for NeoPixel support.   NeoPixels are the WS2812 LEDs with 8 bits per red/green/blue channel.  They are very cheap and fun to use.  However, they can't be "bit banged" from JavaScript.   A NeoPixel data signal is a stream of 24 bits (8 bits per channel x 3 channels) with an approx 1.2 usec duration of pulse duration per bit ...  Thankfully, C code has been widely circulated to achieve this task ...

What would be ideal would be a static class/function that exposed NeoPixel control from the JavaScript environment.   An example of a suggested API might be:

neopixelWrite(pin, array)

where the array is an array of integers defining the color channels.   For example if you had 4 Neopixels in a chain, one might code:

```
neopixelWrite(4, [0xff, 00, 00, // 1st led
00, 0xff, 00, // 2nd led
00, 00, 0xff, // 3rd led
0xff, 0xff, 0xff // 4th led
]);
```

As an alternative to native embedding in Smart.js, perhaps we could design a "plugin" mechanism where C language written modules could be "plugged into" Smart.js environment?  For maximum score, these would be binary files added to the file system as opposed to having to relink/flash the device.

alashkin, 2016-04-15:
Is this issue still actual?
This is a feature request, so could you give me more background on why do you need that? What is the project, and how that functionality will be used?

realzeus, 2016-04-15:
Yes, for me it's still actual.
I use it for visual advertisement, controlled over Internet.
I'm developing a solution to show messages to the drivers behind a car.
Messages are shown using NeoPixels, controlled (now by Arduino), which in turn is controlled by ESP8266-1, which is connected to Smartphone using Wifi.

Bluebie, 2016-04-15:
I would make good use of this. WS2812 has become the defacto standard protocol used by nearly every small digital LED product, from many different vendors and in many different form factors.

bytemage, 2016-05-07:
I too would like to control NeoPixels via an ESP and the Mongoose Firmware.
No commercial background, just tinkering for now :)

bytemage, 2016-05-09:
There is a light weight library to drive the NeoPixels.
https://github.com/cpldcpu/light_ws2812

Would be awesome if you could include it in Mongoose.

bytemage, 2016-05-26:
Any word on this? I have many ideas for little **things** made out of ESPs and WS2812 LEDs all connected over the **internet**, and I'ld love to use the Mongoose IoT Firmware for them.

cpq, 2016-05-30:
At the moment, WS2812 support is lower priority for us, since we have more pressing issues to be implemented / solved - our resources are limited and we have to prioritize wisely.
If there is a commercial project that involves WS2812 - that would bump the priority up. Otherwise, we're not going to look into it until the end of July.

Pull requests are welcome, as usual.

bytemage, 2016-05-30:
While I'm just tinkering, @realzeus seems to be interested in it for a commercial project. But I totally get that you have to prioritize. Thanks for the response.

bytemage, 2016-07-12:
Hey there, just wanted to check back an make sure you don't forget about this ;)

I've been using the Arduion IDE and Blynk to make some remote controlled NeoPixel light effects for now. But a JS based onboard interface would be sweet, and would be a major step to it actually becoming an actual product and maybe even a commercial project.

cpq, 2016-07-13:
@bytemage Hi Simon! Could you elaborate on it please? Is there a concise description of the project - how it works, where it is used, and how JS can improve things, please?

bytemage, 2016-07-20:
Oh, sorry ... So, I've built my first prototypes of "light in a jar", being NeoPixels and and ESP8266 in an jar. Currently I'm using Blynk and the basic ArduinoIDE firmware for the ESP to control the light effects but I'ld like to us Mongoose as I'm working with JS on a daily basis and would like to use it to build a proper configuration panel that is served directly from the ESP, instead of controlling it via an App and a server. Mostly because for Blynk to work the ESP needs an internet connection, and while I'm now simply using my mobile hotspot on my phone, for a proper "product" it needs to work the "mongoose way". Starting with AP mode for the initial setup and then connecting to a user defined WLAN when provided. Currently I have to set specific WLAN credentials before compiling and flashing. That could be extended to be more flexible, but why do that when Mongoose already has such a great setup process. It's just missing the NeoPixel control code ;)

cpq, 2016-07-20:
Why don't some of you guys who's familiar with NeoPixel (none in our engineering team is) just send a PR for it?

cpq, 2016-07-20:
And, by the way, take a look how easy is to build a firmware which has: a) filesystem b) configuration management c) networking, with a single `make` command, no toolchain installation required:

https://github.com/cesanta/mongoose-iot/tree/master/fw/examples

bytemage, 2016-07-20:
There is a light weight library to drive the NeoPixels and other RGB LEDs.
https://github.com/cpldcpu/light_ws2812

I don't think you need to know much about the NeoPixels to include that library, but you do need to know how to integrate it into the Mongoose framework. And at that you guys should be much better at than anyone else.

jamser, 2016-11-05:
ws2812 in mongoose iot ok? how can i use javascript make the RGB LEDs with NeoPixel !~

bytemage, 2016-11-05:
Nope, you can't. WS2812 support is not commercially interesting enough to warrant any effort, it seems.

cpq, 2017-03-19:
https://mongoose-os.com/docs/#/js-api/api_neopixel.js/

closed 2017-03-19 by cpq