cesanta/mongoose-os #83: Request: Expose Sockets API

nkolban, 2015-11-28:
This request asks for exposure of raw sockets APIs both as a socket server and a socket client.  While I fully realize that we have HTTP support, I'd like to also ask for native sockets support.  Ideally, this would include both TCP and UDP flavors.

cpq, 2015-12-09:
Neil, are you talking about something like `Net` API in nodejs , https://nodejs.org/api/net.html ?

nkolban, 2015-12-09:
Yes ... I believe so.  Unfortunately I am not yet skilled in Node.js but a quick glance at the APIs seems to show what I would expect from a sockets implementation.

intx82, 2016-03-03:
+1 
it would by nice use some raw network functions

lukevenediger, 2016-03-08:
Related to #98 (UDP Support)

alashkin, 2016-05-13:
We implemented Node-like TCP & UDP API
See here for details:
https://docs.cesanta.com/mongoose-iot/latest/#/javascript-api/udp/
https://docs.cesanta.com/mongoose-iot/latest/#/javascript-api/tcp/

Feel free to try and report bugs ;)

closed 2016-05-13 by alashkin

dude-mansaa, 2018-06-11:
Above links of UDP and TCP apis are broke. Could you please provide the correct links?

cpq, 2018-06-11:
https://cesanta.com/docs/overview/intro.html