cesanta/mongoose-os #84: How to connect https://dashboard.cesanta.com/?

master-q, 2015-12-06:
I'm using 7e198c386dc9ce1d5f502bf57a8eb28fc1bf98b6, and compile it with following:

```
$ sudo apt-get install qt5-default qt5-qmake qtbase5-dev-tools libqt5serialport5-dev
$ cd smart.js/flashnchips
$ qmake --version
QMake version 3.0
Using Qt version 5.5.1 in /usr/lib/x86_64-linux-gnu
$ make
$ ls -l flashnchips
-rwxr-xr-x 1 kiwamu kiwamu 596952 Dec  5 21:34 flashnchips*
$ sudo apt-get install docker.io
$ cd smart.js/platforms/esp8266
$ make
--snip--
/usr/bin/rsync -r --copy-links fs/ /cesanta/smart.js/platforms/esp8266/build/fs/
CP ../../src/js/smart.js -> fs/smart.js
V7 ../../src/js/clubby.js -> fs/clubby.js
V7 ../../src/js/sys_rts.js -> fs/sys_rts.js
V7 ../../src/js/cloud.js -> fs/cloud.js
V7 ../../src/js/swupdate.js -> fs/swupdate.js
V7 ../../src/js/MCXXX_LCD.js -> fs/MCXXX_LCD.js
V7 ../../src/js/I2C.js -> fs/I2C.js
V7 ../../src/js/demo.js -> fs/demo.js
V7 ../../src/js/fileservice.js -> fs/fileservice.js
  MKSPIFFS fs -> firmware/0x70000.bin
adding files in directory /cesanta/smart.js/platforms/esp8266/build/fs
a MC24FC.js
a sys_rts.js
a MC24FC_test.js
a I2C.js
a fileservice.js
a clubby.js
a MPL115A1.js
a demo.js
a MCXXX_LCD.js
a smart.js
a cloud.js
a gpio.js
a user.js
a swupdate.js
a MCP9808.js
a MCP9808_test.js
Image stats: size=49152, space: total=37901, used=23092, free=14809

smartjs footprint breakdown
---------------------------------
               .text          .irom0.text    .rodata        .data          .bss
mongoose       0              28402          0              0              0
v7             0              79578          0              0              0
smartjs        1202           72705          0              0              0
sys            8924           120543         0              0              0
total          10126          301228         0              0              0

Firmware size is 440Kb (of 512Kb), 15% is available
$ ls -l firmware
total 444
-rw-r--r-- 1 root root  34176 Dec  5 22:28 0x00000.bin
-rw-r--r-- 1 root root 367352 Dec  5 22:28 0x11000.bin
-rw-r--r-- 1 root root  49152 Dec  5 22:28 0x70000.bin
```

And write the firmware into my ESP8266:

```
$ ../../flashnchips/flashnchips
$ picocom -b 115200 --omap crcrlf /dev/ttyUSB0
Version: ESP8266-master/20151205-132822/7e198c38
Device id: //api.cesanta.com/d/dev_54241
Device psk: psk_97082
Cloud: //api.cesanta.com 
HELO! Type some JS. See https://github.com/cesanta/smart.js for more info. 

smartjs 28816/972$
```

However the device can't connect https://dashboard.cesanta.com/.

```
smartjs 28936/1224$ Wifi.setup('aterm-bbfadc-g','3da220e879087')
smartjs 28816/1104$ Wifi.ip()
"192.168.179.3"
smartjs 28816/1104$ Wifi.status()
"got ip"
smartjs 28936/1224$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
fun 0x40237bec sp 0x3fffdfc0 limit 0x3fffdfa0 left 32

Trap 9: pc=0x402691ce va=0x1
Dumping core to debug output
rebooting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 29672, room 16 
tail 8
chksum 0x3f
load 0x3ffe8000, len 2480, room 0 
tail 0
chksum 0x3b
load 0x3ffe89b0, len 1980, room 8 
tail 4
chksum 0x8e
csum 0x8e
don't use rtc mem data
r�Scanning nets 

Version: ESP8266-master/20151205-132822/7e198c38 
Device id: //api.cesanta.com/d/dev_54241 
Device psk: psk_97082 
Cloud: //api.cesanta.com 
HELO! Type some JS. See https://github.com/cesanta/smart.js for more info. 

smartjs 28680/576$ Cannot find known network, use Wifi.setup 
```

What's wrong?

cpq, 2015-12-06:
Quick question - did you register your device (ID & password) on https://dashboard.cesanta.com ?

master-q, 2015-12-10:
Yes. I registered them.

I'll try https://github.com/cesanta/smart.js/releases instead of my own building.

master-q, 2015-12-10:
I got same Device id and Device psk with Smart.js Beta version. Should I re-assign my own id and psk into the firmware?

```
Version: ESP8266-master/20151205-132822/7e198c38 
Device id: //api.cesanta.com/d/dev_54241 
Device psk: psk_97082 
Cloud: //api.cesanta.com 
HELO! Type some JS. See https://github.com/cesanta/smart.js for more info. 

smartjs 28680/576$
```

master-q, 2015-12-10:
I got success with Smart.js Beta version. 

```
$ unar flashnchips-win.zip
$ cd flashnchips-win/firmware/ESP8266/Smart.js
$ esptool.py --baud 115200 --port /dev/ttyUSB0 write_flash 0x00000 0x00000.bin 0x11000 0x11000.bin 0x6d000 0x6d000.bin
$ picocom -b 115200 --omap crcrlf /dev/ttyUSB0
Device id: //api.cesanta.com/d/dev_84227 
Device psk: psk_36751 
Cloud: //api.cesanta.com 
HELO! Type some JS. See https://github.com/cesanta/smart.js for more info. 

smartjs 29032/2268$ Wifi.setup('XXXXXXXX','YYYYYYYYYY')
true
smartjs 29208/2376$ Wifi.ip()
192.168.179.3
smartjs 29016/2280$ Wifi.status()
got ip
smartjs 29112/2280$ demo()
Initializing demo data source 
Starting temperature push demo in background, type Debug.mode(1) to see activity 
undefined
```

I think the git repo version has some error to connect https://dashboard.cesanta.com/.
And I think I should assign own ID into the firmware. How to use flashnchips's --generate-id option?

[image]
[image]
[image]

cpq, 2016-01-03:
Thank you @master-q !

closed 2016-01-03 by cpq