cesanta/mongoose-os #85: Need some patches to build smart.js/platforms/esp8266?

master-q, 2015-12-10:
I found following error on building smart.js/platforms/esp8266.

```
$ pwd
/home/kiwamu/src/smart.js/platforms/esp8266
$ make
make: Entering directory '/home/kiwamu/src/smart.js/platforms/esp8266'
WARNING: Your kernel does not support memory swappiness capabilities, memory swappiness discarded.
make: Entering directory `/cesanta/v7'
make: Nothing to be done for `v7.c'.
make: Leaving directory `/cesanta/v7'
make: Entering directory `/cesanta/smart.js/platforms/esp8266'
Makefile.build:13: ../../../common/platforms/esp8266/common.mk: No such file or directory
make: *** No rule to make target `../../../common/platforms/esp8266/common.mk'.  Stop.
make: Leaving directory `/cesanta/smart.js/platforms/esp8266'
```

Is following patch needed for it?

```diff
diff --git a/platforms/esp8266/Makefile.build b/platforms/esp8266/Makefile.build
index 2f5d9cb..5db8134 100644
--- a/platforms/esp8266/Makefile.build
+++ b/platforms/esp8266/Makefile.build
@@ -7,8 +7,8 @@
 # SSL: if set to "Krypton" enables SSL with the krypton lib
 #
 
-REPO_ROOT ?= ../../..
-COMMON_ESP_DIR ?= $(REPO_ROOT)/common/platforms/esp8266
+REPO_ROOT ?= ../..
+COMMON_ESP_DIR ?= $(REPO_ROOT)/platforms/esp8266
 
 include $(COMMON_ESP_DIR)/common.mk
```

However, I think more patches needed to build it.

cpq, 2015-12-10:
When we export, we still assume private directory structure.
Max, can you think about the way to handle this and make our sources built on public repos, please?

mmikulicic, 2015-12-10:
the script already handled that but it has been broken by a makefile refactoring.

master-q, 2015-12-10:
O.K. I would like to wait the end of refactoring. Thanks.

closed 2015-12-10 by master-q