cesanta/mongoose-os #86: WebDAV on OSX writes zero-length files

aperepel, 2015-12-24:
Latest smartjs firmware on ESP8266 (esp12e board). The Flashnchips tool uploads the js files correctly and I can eval them.

However, if I connect (as a guest) over WebDAV and try to copy/paste the file, it does something (LED blinks), but never flushes the transmit buffers. As a result, files are created, but always zero-length.

cpq, 2015-12-28:
Thanks. Please elaborate on what environment you're using - what DAV client, etc.

aperepel, 2015-12-28:
This is OSX Yosemite. Using a standard built-in WebDAV client (Go to server menu). Connecting as Guest, no password.

You can perform a simple test by navigating to /Volumes/<address> and then touch and edit a file in the root of esp12e filesystem. The file gets created, but content is not written. Drag and drop has similar effects.

cpq, 2016-02-04:
We've done a bunch of research and testing - and decided to ditch DAV as a primary way for file management. It's too hard to make it work smoothly on several platforms.
We're going to use other means, e.g. implementing RESTful API and hooking Web based IDE.

closed 2016-02-04 by cpq

adamclerk, 2016-02-24:
Can you please update your documentation saying such. I spend the better part of 2 hours wrestling with WebDAV and just started uploading the files via Flash&Chips instead.