cesanta/mongoose-os #87: Leverage Esp8266's SmartConfig

aperepel, 2015-12-24:
Hi, I noticed that the beta/dev version of the firmware doesn't use SmartConfig. E.g. after I configured SSID and password, it should be using it upon reboots unless I explicitly erase it. What I see is the device goes into an AP mode on every reboot.

Maybe it's just the dev version? I'm new to smartjs and still navigating around. Noticed there is potentially a place to save those in json config, but wasn't the point of SmartConfig to avoid putting it anywhere in clear-text (e.g. it goes directly into EEPROM).

The process is:
* boot up esp8266 in station mode (it sniffs on the SSID packets)
* software broadcasts specially formatted packets for esp8266 to have it configured automatically. E.g. this one works great: https://play.google.com/store/apps/details?id=com.cmmakerclub.iot.esptouch&hl=en
* esp8266 picks up SSID/pass to connect to and saves creds. They will persist upon reboots until erased via API

Just in case, here's a sample C code for ESP8266 which leverages SmartConfig and responds to a wired button acting as a WiFi credentials reset.

```
#include <ESP8266WiFi.h>
#include <Esp.h>
#include <Button.h>

#define BUTTON_PIN D2
#define PULLUP true
#define INVERT true
#define DEBOUNCE_MS 20

Button *btn;

void setup() {

  btn = new Button(BUTTON_PIN, PULLUP, INVERT, DEBOUNCE_MS);

  Serial.begin(9600);
  WiFi.mode(WIFI_STA);
  Serial.println();

  // Start SmartConfig if necessary
  if (WiFi.SSID() == "") {

    Serial.println("Beginning SmartConfig");
    WiFi.beginSmartConfig();

    while (WiFi.status() != WL_CONNECTED) {
      delay(1000);
      Serial.print(".");
      if (WiFi.smartConfigDone()) {
        Serial.println();
        Serial.println("SmartConfig completed");
        WiFi.stopSmartConfig();
        delay(2000);
        break;
      }
    }
  } else {
    Serial.println(String("Using a saved SSID: ") + WiFi.SSID());
  }

  Serial.print("Connecting..");
  WiFi.begin();
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  Serial.println("Ok");

  Serial.println(String("My IP: ") + WiFi.localIP().toString());
}

void checkResetButton() {
  btn->read();
  if (btn->wasReleased()) {
    Serial.println("Resetting WiFi credentials");
    WiFi.disconnect();
    ESP.restart();
  }
}

void loop() {
  checkResetButton();

}
```

cpq, 2015-12-26:
Thanks Andrew.

We do not plan to use smartconfig. Smart.js uses it's own configuration infrastructure, where configuration is saved in the json files, and also persists through reboots.

The way to pass wifi configuration for Smart.js would be manual via AP mode web UI setup, or by copying appropriate json config file to the FS.

aperepel, 2015-12-26:
Thanks, could you please add a section/paragraph in a doc describing the workflow? E.g. post-flashing the firmware, which config file should be added (or modified for that matter)? I'd imagine this will be a very common request, given that the latest dev release 'forgets' the config on every reboot by default.

cpq, 2016-02-04:
Sure, that should be relevant:

https://www.cesanta.com/developer/smartjs#_firmware_startup_process

closed 2016-02-04 by cpq