cesanta/mongoose-os #89: Where is Wifi.mode in Beta?

heymind, 2016-01-08:
I want to use my esp8266 in Ap mode with smart.js.When I read the document,I know I should use `Wifi.mode` method.But ....

Look at the screenshot.How can I  setup for a wireless access point?

[image]

snsekar, 2016-01-11:
I am also facing same issue. Please help me

alashkin, 2016-01-14:
@heymind 
@snsekar 

Hi! 
Do you build smart.js from sources or use pre-built binaries?
In case of building from source, you  can try to use HEAD of this repo.

It still doesn't have `Wifi.mode`, but Wifi mode can be specified via config (`conf_sys_defaults.json`)
NOTE: it it better to create your own `conf.json` next to `conf_sys_defaults.json` and put overriden values there.
Btw, current HEAD by default starts in AP mode, in order to use STA mode the following section in configuration must be filled. 

```
    "sta": {
      "enable": false,
      "ssid": "",
      "pass": ""
    },
```

It you want pre-built binaries, more-or-less fresh can be download here:
https://backend.cesanta.com/devel/win_fw_20160104_224731.zip
https://backend.cesanta.com/devel/mac_fw_20160104_224731.zip

snsekar, 2016-01-14:
@alashkin i used windows pre-build binaries from here https://github.com/cesanta/smart.js . Currently i am out of office now. I wont be able to check couple of days. I will update here when i check

closed 2016-02-05 by heymind