cesanta/mongoose-os #90: WebSocket client not working

snsekar, 2016-01-12:
Hi,
Today i flashed windows build from here https://github.com/cesanta/smart.js on my esp8266-01. I am trying to run WebSocket example given here https://www.cesanta.com/developer/smartjs#_websocket

But i am getting Exec error: {"message":"error creating the connection"}
[image]

However i am able to connect to Wifi successfuly and Http.request is working fine.

mmikulicic, 2016-01-12:
@alashkin reproduced with HEAD

mmikulicic, 2016-01-12:
@alashkin sj_mongoose_websocket just passes the raw arg to mongoose low level tcp connect.
It should use http connect

snsekar, 2016-01-13:
Thanks for looking into this issue. If WebSocket client works fine, this will be killer feature for our project.

mmikulicic, 2016-01-13:
Hi @snsekar,

The latest master branch has working websocket support. Could you please try it out?

snsekar, 2016-01-14:
Wow great...i am out of office now. I wont be able to check couple of days. I will update here when i check

marsteini, 2016-01-17:
Hi, I'm experiencing the same problem. Is there a binary build of the current branch?

mmikulicic, 2016-02-18:
The latest binary build is available at https://github.com/cesanta/smart.js under the `Current development Smart.js firmware builds` bullet.

This issue has been fixed. Feel free to reopen if you have any problems.

closed 2016-02-18 by mmikulicic

snsekar, 2016-02-25:
Thanks for fixing this issue. I will check

snsekar, 2016-03-04:
I checked the build mentioned by @mmikulicic . WebSocket Client working now.