cesanta/mongoose-os #91: AWS IoT support

snsekar, 2016-01-15:
Hi Smart.js team,
Smart.js is great device firmware for web communication.  AWS IoT is one of the emerging AEP. I am thinking to combine best of both worlds. Is it possible to support AWS IoT with smart.js. Is it big job? or can be done in reasonable time? (within couple of months with 2 developers). Already some implementations are there https://aws.amazon.com/iot/getting-started/#kits

I am assuming this can be achieved in two ways. Embedded C (or) Javascript

They are providing device SDKs https://aws.amazon.com/iot/sdk/

Javascript variant 
https://github.com/aws/aws-iot-device-sdk-js

Embedded C variant
https://github.com/aws/aws-iot-device-sdk-embedded-C

Which approach will be best to proceed ? or Do you have any plans for AWS IoT ?

cpq, 2016-01-16:
Yes, AWS interoperability looks interesting.
Do you work on the specific IoT deployment using AWS IoT ?

snsekar, 2016-01-18:
Yes. Currently we are researching about backend platform which is suitable for our use case. Our use case is similar to home automation products. We need to support millions of devices with lower cost. Also platform needs to stable , matured and million scale with lower TCO. AWS IoT seems promising option for us. If we finalise AWS IoT platform then we will start work on this.

cpq, 2016-12-27:
AWS IoT is implemented. 
https://github.com/cesanta/mongoose-iot/tree/master/fw/examples/c_mqtt

closed 2016-12-27 by cpq

snsekar, 2016-12-28:
@cpq Wow...Thank you so much...I will ask my team to try out

cpq, 2016-12-28:
Please do.
Note that our solution, including cryptochip support for secure SSL key storage, has been reported by our Amazon partners on re:Invent in Vegas recently (see https://mongoose-iot.com/iot_security.html).
This is the ONLY working ESP8266 + AWS IoT solution we're aware of.