cesanta/mongoose-os #92: Wifi changed state (Trap:9 pc=0x4024eda3 va=0xa Dumping core to debug output rebooting)

drrest, 2016-02-06:
Hi! My code is:

```
Wifi.setup("[ssid]","[psk-key]");

function do_socket() {
	var ws = new WebSocket('ws://[IP]:8800');

		ws.onopen = function(ev) {
		  ws.send('HELLO=I_Guest');
		};

		ws.onmessage = function(ev) {
		  console.log('< ', ev.data);
		};
}

Wifi.changed(function(state) {
  print('Debug: ', state);
  if(state == 3) {
	do_socket();
  }
});
```

After Wifi state changed to ( 2 )  all crashed and i receive:

> Trap 9: pc=0x4024eda3 va=0xa
> Dumping core to debug output
> rebooting

What i do wrong?

Thank you

mmikulicic, 2016-02-06:
hi, which version of the fw are you running?

BTW, it's better to use Wifi.ready instead of Wifi.changed. You can configure WiFi with the web configuration UI. 

Wifi.ready will invoke the callback immediately if the WiFi is already connected. This avoids a race condition

drrest, 2016-02-07:
"fw_id":"20160128-174140/ui@f20e0990","fw_timestamp":"2016-01-28T17:41:40.209322","fw_version":"20160128174140","arch":"esp8266"

Also i can 't read DHT11 values (i check on several boards.((( DHT11.read(xx) - have result - string "null"

alashkin, 2016-04-14:
Wifi.changed crash fixed in  https://github.com/cesanta/smart.js/commit/580e2e5c06bd2283fb5dd42af68a06b0c6b78930

As for DHT11, its support was discontinued

closed 2016-04-14 by alashkin