cesanta/mongoose-os #93: Support 512kb modules

ssserebrov, 2016-02-10:
Is it correct that beta2+ builds supports only esp8266 modules with 1 megabyte and more?
Only beta build is applicable to esp8266 512kb?

mmikulicic, 2016-02-10:
Yes we temporarily dropped support for 512kb flash. You can change the FS size in the makefile and make it fit.

We default to a bigger FS because our main use case is to actually put stuff on the filesystem, such as our configuration web ui html, and some JS file implementing smartjs functionality. 

You can fiddle a bit with the makefile and remove the features you don't need.

We also focus mostly on OTA, which requires bigger flashes anyway.

alashkin, 2016-04-14:
Support of ESP modules with 512Kb flash was discontinued.

closed 2016-04-14 by alashkin