cesanta/mongoose-os #95: Exec error: {"message":"[OS] is not defined"}

pyromaticx, 2016-02-14:
Hello

Just getting started playing around with this framework so I apologize if this is a mistake. 

When running the blink.js example the board crashes and resets after about 20 interations of setTimeout, and also when used in a simple for loop exceeding 20 iterations

```
fun 0x4022b878 sp 0x3fffe090 limit 0x3fffdfa0 left 240

Trap 9: pc=0x4027d4fa va=0x1
Dumping core to debug output
```

I understand that the ESP8266 has a watchdog timer. When I try to access the `OS.wdt_feed()` method i receive the error `Exec error: {"message":"[OS] is not defined"}`

I tried to locate the module referenced in the documentation, but could not. This is the only mention that I can tell of this module.

Was this changed in the last commit - "Good bye osdep.h, you won't be missed"?

```
"ro_vars":{"fw_id":"20160128-174140/ui@f20e0990","fw_timestamp":"2016-01-28T17:41:40.209322","fw_version":"20160128174140","arch":"esp8266","mac_address":"5ECF7F0C3F90"}} 
```
Using the NODEMCU v0.9 board

mmikulicic, 2016-02-14:
Hi,

We changed OS to Sys a while ago. It's unrelated to the recent internal adep.h change.

As for the crash, it would be great if you could attach the coredump and the exact firmware image you are using. The coredump can be obtained by changing the debug output to uart0 either via the web configuration ui or by invoking Debug.mode(1) at the prompt.

pyromaticx, 2016-02-15:
Here is the debug output. 
[stacktrace.txt](https://github.com/cesanta/smart.js/files/130031/stacktrace.txt)

I am using the beta2 firmware. 

The code being executed is:

```
function blink(pin) {
  var pin = pin || 0;
  var level = GPIO.read(pin);
  GPIO.setmode(pin, 0, 0);
  GPIO.write(pin, !level);
  console.log(GC.stat()); // added after crashing to test stats
  setTimeout(function() { blink(pin); }, 500);
}
```

What I did notice from logging `GC.stat()` during each iteration that `"sysfree"` is < 1500 before crash. Is this little bit of code depleting all free memory before it can be freed / garbage collected?
first log:
```
{"owned_max":8,"owned":4,"astsize":19,"funcncell":16,"funcnfree":3,"propncell":24,"propnfree":2,"objncell":12,"objfree":6,"struse":1814,"strres":2284,"jsfree":168,"jssize":9556,"sysfree":29448}
```

Last log:
```
{"owned_max":10,"owned":7,"astsize":0,"funcncell":16,"funcnfree":2,"propncell":24,"propnfree":2,"objncell":12,"objfree":0,"struse":1814,"strres":2284,"jsfree":80,"jssize":34676,"sysfree":1472} 
fun 0x4022b878 sp 0x3fffe090 limit 0x3fffdfa0 left 240

Trap 9: pc=0x4027d4fa va=0x1
Dumping core to debug output
rebooting
```

mmikulicic, 2016-02-15:
Hi,

yeah, I'm sorry, this is a known memory leak which is still present in the Beta2 but solved after it.

You can grab the developer build from https://backend.cesanta.com/devel/smartjs-esp8266-last.zip

---
For the record, a workaround to make it work with the Beta2 release.
 **you don't need this hack with later builds**:

```js
function blink(pin) {
  var pin = pin || 0;
  function tick() {
    var level = GPIO.read(pin);
    GPIO.setmode(pin, 0, 0);
    GPIO.write(pin, !level);
    console.log(GC.stat()); // added after crashing to test stats
    setTimeout(tick, 500);
  }
  tick();
}
```

pyromaticx, 2016-02-16:
Thanks, that solved it. I also noticed you updated the documentation for `Sys.wdtfeed()` Keep up the great work, really liking the framework.

closed 2016-02-16 by mmikulicic