cesanta/mongoose-os #96: Beta2: all broken after Configuration

olterion, 2016-02-19:
Hello,

I have a ESP-01 with 1 Mbyte storage.
With the Flash'n'Chips I flased the beta2 firmware to my ESP-01.
After disconnecting GPIO0 from GND and rebooting I can make everything in the console and I can connect to the Wifi SMARTJS_....... and open the Configuration Console in my browser. After entering my Wifi settings of my router and saving the settings in the console gets this:

smartjs 29648/156$ 
 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 1688, room 16 
tail 8
chksum 0xa9
load 0x3ffe8000, len 784, room 0 
tail 0
chksum 0x9e
csum 0x9e

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
r�ùl�d�Init error: {"message":"cannot open [sys_init.js]"}

After this no connection to my router is done and also I can't do any command in the console. For example I type "File.list('.')" the result of my ESP-01 is "undefined"

smartjs 39632/352$ File.list('.')
undefined

So, what can I do?

With the console of windows and the --log fnc.log --V=4 --gui options I created a lofile of the flashing process. Maybe there is something wrong?!
[http://pastebin.com/9NFcpWxQ](url)

mmikulicic, 2016-02-19:
Could it be flash parameters? We never tried it with a 1MB flash. Do you know which flash chip it is?

olterion, 2016-02-19:
Hello,

before I had a ESP-01 with 512kByte flash. No I have ordered this one with 1 MByte flash special for smart.js

This Board with 1 MByte flash has the flash chip 25Q80A.

Which board is a good one to start with smart.js?

mmikulicic, 2016-02-19:
We test mostly on esp12-E (which includes the nodemcu v1 (aka v2) board).
The nodemcu board is very handy for prototyping; it includes the usb<->tty uart with the right wiring for easy flashing. 
For production we focus on esp12-E.
Both have 4MB flash chips.

We focus on 4MB chips because it allows to perform OTA updates which are a very important feature.

That said, we'd like to make it work on the 1MB flash ESP-01 as well.

Which firmware version are you using ? How are you flashing it, with esptool.py or with FnC ?

Can you try this firmware: https://backend.cesanta.com/devel/smartjs-esp8266-last.zip
And try to flash it with https://github.com/cesanta/fnc/releases/latest ?

mmikulicic, 2016-02-19:
For the record, slightly off topic, but perhaps useful to put previous comment in context:

Our build should work with 1MB flash size. There should be no reason for it to not work if it fits.
We need 4MB for OTA because 

1. we need > 512k for the FW+FS and we need room for at least two version of the FW at the same time 2. even if it fits in <1M, due to the way OTA works, we need to put the version in 1MB windows in the memory mapped address space, i.e. each FW version must use a 1MB section.

olterion, 2016-02-20:
With the last Flash'n'Chips (20160216-132230/0.8.1@b97444d6) and the last firmware (smartjs-20160215095410) is the same behavior.
After the first reboot everything is fine, after the second reboot the sys_init.js cannot be opened.

mmikulicic, 2016-02-20:
@alashkin, I have a vague memory that we had a similar problem before autodetecting the flash type from the flash ID. What would the right flash params be in this case?

mmikulicic, 2016-02-20:
@olterion, could you please try to pass `0220` as `esp8266-flash-params` flag to FnC ?
You can set flags from gui as well, under the `Advanced settings` menu.

IIRC, the first byte is flash mode 0x2 -> DIO, while the second encodes the flash size.
00 -> 4Mb == 512 KB
20 -> 8Mb == 1MB
30 -> 16Mb == 2MB
40 -> 32Mb == 4MB

If it doesn't work, please try other combinations such as `0020`.

Could you also please run the `esptool.py flash_id` tool and paste the output? This way we could add this flash type to our autodetection database.

rojer, 2016-02-20:
i don't think flash params are at issue here. 
nevertheless, can you please post log of the flashing process from `Help -> Log`?

olterion, 2016-02-20:
@mmikulicic 
I tried the esp8266-flash-params with 0220 and 0020. After the flash process the Flash'n'Chips program stopped.
So I tried to enter 0x0220 and 0x0020 in the esp8266-flash-params. But in both cases the behavior after the second reboot was the same than before (sys_init.js not found)

I installed the esptool and here is the output of my flash_id:
Manufacturer: e0
Device: 4014

@rojer 
And here is the log of the flashing process:
http://pastebin.com/fjkZzN5m

rojer, 2016-02-20:
flash size is detected correctly:
```
1 Flash chip ID: 0xe0 0x40 0x14
1 Flash size: 1048576
1 Setting flash params to 0x220
```

olterion, 2016-03-02:
Now I'm really confused about this software.
I bought exactly the recommended ESP12-E board with a 4MB flash.
This is what the flash_id from esptool says:
Manufacturer: e0
Device: 4016

and this is in the log:
1 Flash chip ID: 0xe0 0x40 0x16
1 Flash size: 4194304
1 Setting flash params to 0x220

But everything is the same as the other chip.
When I flash the ESP12-E with FNC 0.8.1 and the lastest smartjs firmware I get after the first reboot the message
`Init error: {"message":"cannot open [sys_init.js]"}`

So now I have the CH340 serial chip insted of the ftdi chip before.
Also I tried to flas it with a linux and the FNC for linux.
But there is the same behavior.

I really don't know why I get everytime the same failure with different PCs and different operating systems.

rojer, 2016-03-02:
please try https://backend.cesanta.com/devel/smartjs-esp8266-last.zip  -
this is the latest firmware

amid0, 2016-03-03:
Hi rojer.
I attaching dumps after our today conversation in chat

[dumps.zip](https://github.com/cesanta/smart.js/files/155978/dumps.zip)

rojer, 2016-03-03:
huge thanks to @amid0 for helping debug.
together we found that on subsequent boots something goes wrong with the FS such that it can (no longer) be mounted - spiffs error 10025, failure to read config, core dump. that's on newer firmware versions, that actually check the return code from reading config... :)
on older firmware it manifests as inability to load .js
amid0 took esptool dumps of the FS - note that according to loader his flash is 1MByte / 8 Mbits, but dump is 4MB just in case. fs.dump is take at address 0xe0000, full dump is from 0.
from my side, i'm attaching the exact firmware that was flashed.
[smartjs-esp8266-last.zip](https://github.com/cesanta/smart.js/files/155986/smartjs-esp8266-last.zip)
it was flashed with F&C so we can be sure it was intact on first boot (and reportedly, first boot works fine).
the second boot with debug on produces this:

```
: sta(5c:cfs_mount             Mount 131072@E0000
fs_mount             SPIFFS_mount failed: -10025
load_config_file     === Loading conf_sys_defaults.json
set_errno            spiffs error: -10000
init_device          Failed to load config defaults
sjs_init             init_device failed```

amid0, 2016-03-06:
Hi guys. Is the any updates on issue?

cpq, 2016-03-10:
@amid0 , @rojer will be able to help with it in ~2 weeks time, so far there is no update.

rojer, 2016-03-17:
@alashkin Alex, it seems that this is specific to modules with 8Mb/1MB flash chips (ESP-07 in particular).
_something_ happens to FS. do you know what could it be?

it occurs to me that OTA will not work on modules with < 16Mb/2MB flash because of windowing... but in theory that should not happen before update. can you think of anything else that would go wrong on 1MB flash?

rojer, 2016-03-18:
ok, i think we have this figured out.
please try flashing the latest FW with the latest F&C (0.8.2)
https://backend.cesanta.com/devel/smartjs-esp8266-last.zip
https://github.com/cesanta/fnc/releases/tag/0.8.2

reopen if this doesn't fix the problem.

closed 2016-03-18 by rojer

amid0, 2016-03-18:
Ok. It seems working for me.
Well done guys.
Thank you.
Now I can continue to learning SmartJS.