cesanta/mongoose-os #97: Boot fails after flashing a local build of smart.js

lukevenediger, 2016-02-21:
Hi there,

I followed the instructions on how to flash the esp8266 and am getting a "system param error" when I reboot after flashing. Here are the instructions: https://github.com/cesanta/smart.js/tree/master/smartjs/platforms/esp8266

I flashed all four binary images:
`esptool.py --baud 115200 --port /dev/cu.wchusbserial1410 write_flash 0x00000 0x00000.bin 0x1000 0x01000.bin 0x11000 0x11000.bin 0xe0000 0xe0000.bin`

Here's the dev board I'm using: http://www.aliexpress.com/item/ESP8266-serial-WIFI-Witty-cloud-Development-Board-ESP-12F-module-MINI-nodemcu/32566502491.html

This is the error I get:

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

```
load 0x40100000, len 1688, room 16 
tail 8
chksum 0xa9
load 0x3ffe8000, len 784, room 0 
tail 0
chksum 0x9e
csum 0x9e

rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   4 Mbit
Flash Mode:   QIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
system param error
41 72 72 61 79 00 f9 ff 60 c2 27 40 00 00 ff ff f8 cf 27 40 47 00 a1 e9 45 72 72 6f 72 00 f9 ff 
e4 c2 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 34 0f 23 40 00 00 f4 ff 
28 d0 27 40 47 00 a1 e9 51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 00 00 00 00 47 00 a1 e9 
26 00 00 00 00 00 f3 ff 78 c2 27 40 00 00 ff ff 40 d0 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff 
6c c2 27 40 00 00 ff ff 10 d0 27 40 47 00 a1 e9 17 00 00 00 00 00 f3 ff 78 c2 27 40 00 00 ff ff 
00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 34 0f 23 40 00 00 f4 ff a0 d0 27 40 47 00 a1 e9 
51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 00 00 00 00 47 00 a1 e9 26 00 00 00 00 00 f3 ff 
90 c2 27 40 00 00 ff ff b8 d0 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff 84 c2 27 40 00 00 ff ff 
88 d0 27 40 47 00 a1 e9 16 00 00 00 00 00 f3 ff 90 c2 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 
00 00 00 00 00 00 fa ff 34 0f 23 40 00 00 f4 ff 18 d1 27 40 47 00 a1 e9 51 00 00 00 00 00 f3 ff 
00 00 00 00 00 00 f0 3f 00 00 00 00 47 00 a1 e9 26 00 00 00 00 00 f3 ff a8 c2 27 40 00 00 ff ff 
30 d1 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff 9c c2 27 40 00 00 ff ff 00 d1 27 40 47 00 a1 e9 
11 00 00 00 00 00 f3 ff a8 c2 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
34 0f 23 40 00 00 f4 ff 90 d1 27 40 47 00 a1 e9 51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 
00 00 00 00 47 00 a1 e9 26 00 00 00 00 00 f3 ff c0 c2 27 40 00 00 ff ff a8 d1 27 40 47 00 a1 e9 
5f 00 00 00 00 00 f3 ff b4 c2 27 40 00 00 ff ff 78 d1 27 40 47 00 a1 e9 08 00 00 00 00 00 f3 ff 
c0 c2 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 34 0f 23 40 00 00 f4 ff 
08 d2 27 40 47 00 a1 e9 51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 00 00 00 00 47 00 a1 e9 
26 00 00 00 00 00 f3 ff d8 c2 27 40 00 00 ff ff 20 d2 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff 
cc c2 27 40 00 00 ff ff f0 d1 27 40 47 00 a1 e9 10 00 00 00 00 00 f3 ff d8 c2 27 40 00 00 ff ff 
00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 34 0f 23 40 00 00 f4 ff 80 d2 27 40 47 00 a1 e9 
51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
34 0f 23 40 00 00 f4 ff b0 d2 27 40 47 00 a1 e9 51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 
c8 d2 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff fc c2 27 40 00 00 ff ff 00 00 00 00 47 00 a1 e9 
26 00 00 00 00 00 f3 ff e4 c2 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
18 ca 22 40 00 00 f4 ff 10 d3 27 40 47 00 a1 e9 51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 00 00 
f8 d2 27 40 47 00 a1 e9 9a 00 00 00 00 00 f3 ff f0 c2 27 40 00 00 ff ff 00 00 00 00 47 00 a1 e9 
26 00 00 00 00 00 f3 ff 14 c3 27 40 00 00 ff ff 98 d2 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff 
08 c3 27 40 00 00 ff ff 68 d2 27 40 47 00 a1 e9 05 00 00 00 00 00 f3 ff 14 c3 27 40 00 00 ff ff 
00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 80 b3 22 40 00 00 f4 ff a0 d3 27 40 47 00 a1 e9 
51 00 00 00 00 00 f3 ff 00 00 00 00 00 00 f0 3f 00 00 00 00 47 00 a1 e9 26 00 00 00 00 00 f3 ff 
2c c3 27 40 00 00 ff ff d0 d3 27 40 47 00 a1 e9 9f 00 00 00 00 00 f3 ff 44 07 23 40 00 00 f4 ff 
e8 d3 27 40 47 00 a1 e9 9a 00 00 00 00 00 f3 ff 34 1a 23 40 00 00 f4 ff b8 d3 27 40 47 00 a1 e9 
5f 00 00 00 00 00 f3 ff 20 c3 27 40 00 00 ff ff 88 d3 27 40 45 00 a1 e9 01 00 00 00 00 00 f3 ff 
2c c3 27 40 00 00 ff ff 00 00 00 00 47 00 a1 e9 03 6d 61 78 00 00 fa ff 34 d3 22 40 00 00 f4 ff 
48 d4 27 40 47 00 a1 e9 03 6d 69 6e 00 00 fa ff 00 d3 22 40 00 00 f4 ff 60 d4 27 40 47 00 a1 e9 
60 00 00 00 
c0 dd 27 40 45 00 a1 e9 03 49 32 43 00 00 fa ff 48 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 
00 00 00 00 00 00 fa ff 98 a2 24 40 00 00 f4 ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
c4 a4 24 40 00 00 f4 ff 00 00 00 00 47 00 a1 e9 04 74 72 61 6e 00 fa ff 54 c5 27 40 00 00 ff ff 
00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff e0 a3 24 40 00 00 f4 ff 48 e0 27 40 47 00 a1 e9 
03 74 78 6e 00 00 fa ff 60 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
30 a6 24 40 00 00 f4 ff 78 e0 27 40 47 00 a1 e9 63 6c 6f 73 65 00 f9 ff 6c c5 27 40 00 00 ff ff 
a8 e0 27 40 47 00 a1 e9 26 00 00 00 00 00 f3 ff 84 c5 27 40 00 00 ff ff 18 e0 27 40 47 00 a1 e9 
5f 00 00 00 00 00 f3 ff 78 c5 27 40 00 00 ff ff 00 e0 27 40 45 00 a1 e9 03 53 50 49 00 00 fa ff 
84 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 14 b8 24 40 00 00 f4 ff 
00 00 00 00 47 00 a1 e9 26 00 00 00 00 00 f3 ff c0 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 
00 00 00 00 00 00 fa ff 04 bf 24 40 00 00 f4 ff 20 e1 27 40 47 00 a1 e9 04 73 65 6e 64 00 fa ff 
90 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 78 bd 24 40 00 00 f4 ff 
50 e1 27 40 47 00 a1 e9 63 6c 6f 73 65 00 f9 ff 9c c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 
00 00 00 00 00 00 fa ff 2c ba 24 40 00 00 f4 ff 80 e1 27 40 4f 00 a1 e9 65 00 00 00 00 00 f3 ff 
a8 c5 27 40 00 00 ff ff 08 e1 27 40 47 00 a1 e9 5f 00 00 00 00 00 f3 ff b4 c5 27 40 00 00 ff ff 
c8 e1 27 40 45 00 a1 e9 04 4f 50 45 4e 00 fa ff 00 00 00 00 00 00 f0 3f e0 e1 27 40 45 00 a1 e9 
02 00 00 00 00 00 f3 ff 00 00 00 00 00 00 00 40 f0 e0 27 40 45 00 a1 e9 19 00 00 00 00 00 f3 ff 
c0 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff e4 b5 24 40 00 00 f4 ff 
00 00 00 00 47 00 a1 e9 04 6d 6f 64 65 00 fa ff cc c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 
00 00 00 00 00 00 fa ff 74 b6 24 40 00 00 f4 ff 40 e2 27 40 47 00 a1 e9 70 72 69 6e 74 00 f9 ff 
d8 c5 27 40 00 00 ff ff 70 e2 27 40 45 00 a1 e9 03 4f 46 46 00 00 fa ff 00 00 00 00 00 00 00 00 
88 e2 27 40 45 00 a1 e9 03 4f 55 54 00 00 fa ff 00 00 00 00 00 00 f0 3f a0 e2 27 40 45 00 a1 e9 
03 45 52 52 00 00 fa ff 00 00 00 00 00 00 00 40 10 e2 27 40 45 00 a1 e9 44 65 62 75 67 00 f9 ff 
e4 c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 94 8c 24 40 00 00 f4 ff 
00 00 00 00 47 00 a1 e9 52 00 00 00 00 00 f3 ff f0 c5 27 40 00 00 ff ff 00 00 00 00 45 00 a1 e9 
5f 73 65 72 76 00 f9 ff fc c5 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
14 89 24 40 00 00 f4 ff 00 00 00 00 47 00 a1 e9 a2 00 00 00 00 00 f3 ff 08 c6 27 40 00 00 ff ff 
00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 34 86 24 40 00 00 f4 ff 48 e3 27 40 47 00 a1 e9 
77 72 69 74 65 00 f9 ff 14 c6 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 
00 87 24 40 00 00 f4 ff 78 e3 27 40 47 00 a1 e9 03 65 6e 64 00 00 fa ff 20 c6 27 40 00 00 ff ff 
00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff a8 8a 24 40 00 00 f4 ff a8 e3 27 40 47 00 a1 e9 
73 65 72 76 65 00 f9 ff 2c c6 27 40 00 00 ff ff 18 e3 27 40 45 00 a1 e9 5f 72 65 73 70 00 f9 ff 
38 c6 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa ff 7c 87 24 40 00 00 f4 ff 
00 00 00 00 47 00 a1 e9 77 72 69 74 65 00 f9 ff 44 c6 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 
00 00 00 00 00 00 fa ff e8 87 24 40 00 00 f4 ff 20 e4 27 40 47 00 a1 e9 03 65 6e 64 00 00 fa ff 
50 c6 27 40 00 00 ff ff 00 00 00 00 65 00 a1 e9 00 00 00 00 00 00 fa f
```

Any idea of what's going wrong?

Thanks!

cpq, 2016-02-22:
@alashkin Please take a look.

alashkin, 2016-02-22:
Hello, @lukevenediger 

What flash size is on your chip?
As I see, it is flashed like 512Kb 
`Flash Size:   4 Mbit`
while smart.js requires at least 8Mbit (1MB)

I didn't find info about flash size in your link, but assuming it is still ESP-12, I'd suggest to try to flash with ` --flash_mode dio --flash_size 32m`
i.e.
```
esptool.py --baud 115200 --port /dev/cu.wchusbserial1410 write_flash  --flash_mode dio --flash_size 32m 0x00000 0x00000.bin 0x1000 0x01000.bin 0x11000 0x11000.bin 0xe0000 0xe0000.bin
```

or use our flasher (https://github.com/cesanta/fnc)
  
Please, try and let me know if works/not.

lukevenediger, 2016-02-22:
Hi @alashkin,

This morning I tried using FNC to flash the device instead and managed to get it working (yay!). I had to tell it to flash at a baud rate of 115200. Also, flashing using FNC was at least 4x faster than using esptool.py.

Here's what I used to flash the chip:
`/Applications/FNC.app/Contents/MacOS/FNC --flash-baud-rate 115200 --flash smartjs-esp8266-20160220203212.zip --platform esp8266 --port /dev/cu.wchusbserial1410 `

I was able to flash my NodeMCU and the Witty board I mentioned in the issue. I'm guessing there's some magic happening in FNC that causes it to flash properly. The speed difference is significant too - it's over a minute using esptool.py and about 5 seconds using FNC.

I'll leave you to close this unless you have any questions or want me to try something else.

Thanks,
Luke

alashkin, 2016-02-22:
Hi @lukevenediger 

Well, it everything is OK, and everybody is happy, I have no questions.
What I can offer to try?
Well, you can try ` --flash-baud-rate 921600` instead of `115200`. I usually use that speed.

closed 2016-02-23 by alashkin