cesanta/mongoose-os #100: Example "http server example" isn't working

olterion, 2016-03-03:
Hello,

I have a working smart.js firmware on a ESP12-E with 4MB flash.
At the moment I have the Beta2 smart.js firmware on it.
I want to have a little standalone webserver like in the "http webserver example" shown.
But I can't get it running.
I store the code snippets from https://www.cesanta.com/developer/smartjs#_http_server_example at "http_server_1.js" and "index.html" an load it onto the ESP12-E.
The I type `File.eval('http_server_1.js')` into the FNC console and here is the failure:

`Exec error: {"message":"Cannot bind"}`

When I try the functionality of this project, I can see the html page in the webbrowser under the IP address of my board.
But if I enter a GPIO port and push "Set" on the page, nothing happens. The port voltage doesn't change and the page become blank and the address stays at `board_ip/gpio`.
Also in the FNC console nothing receives.

So, what can be wrong in this case?

TRAHOMOTO, 2016-03-04:
Same problems

mmikulicic, 2016-03-04:
That's because smartjs boots with a webserver on port 80, please use another port.

Sorry, we'll update the doc and show how to disable the embedded web conf web server.

TRAHOMOTO, 2016-03-04:
Yes! You are damn right!

closed 2016-03-04 by alashkin