cesanta/mongoose-os #98: UDP support

lukevenediger, 2016-03-03:
Hi,

Are there any plans to expose Mongoose's UDP support into the smartjs JavaScript runtime?

Thanks!
Luke

cpq, 2016-03-03:
Yes.
Luke, do you have an idea in mind with regard to the API?
Could you elaborate on your use case please?

lukevenediger, 2016-03-03:
Hi,

Yes for sure. My use case is being able to send lightweight messages to a data broker that's connected to Firebase (see https://github.com/lukevenediger/firebase-udp-bridge/ ). I could add support for websockets but there might be other clients wanting to connect to the broker that don't have websocket support.

In terms of the API, I guess I'd like it to be similar to how you send and receive UDP messages in Node.js, using the Socket class (see https://nodejs.org/api/dgram ), but held in a root namespace called UDP.

Here's an idea for a UDP API:

```javascript
/**
 * Send UDP packets to remote hosts. Listen for incoming packets and process them.
 * @singleton
 */
var UDP = {
  /**
   * Send a UDP packet
   * @param {Array[]} bytes the payload
   * @param {String} remoteHost the recipient's IP or hostname
   * @param {Number} remotePort the recipient's listen port
   * @param {Function} onError callback that's fired if an error occurs
   */
  send: function(bytes, remoteHost, remotePort, onError) {},
 
  /**
   * Bind to an address and port and listen for incoming packets
   * @param {String} address the local address to bind to
   * @param {Number} port the local port to listen on
   * @param {Function} onReady callback that's fired when the listener is active
   * @param {Function} onMessage callback that's fired when a new packet arrives
   * @param {Function} onError callback that's fired if an error occurs
   * @returns {Function} a function that when called it will stop listening for traffic
  bind: function(address, port, onReady, onMessage, onError) {
    return function cancelBind() { ... };
  }
};
```

Here's a description of each of the callback functions passed to "bind"

```javascript
/**
 * Called when the socket is ready to accept data
 * @param {String} listenAddress the IP or hostname that the socket is bound to
 * @param {Number} listenPort the local listen port
 */
function onReady(listenAddress, listenPort) {
};

/**
 * Called when a new packet arrives on the socket
 * @param {Array[]} data an array of bytes
 * @param {String} remoteAddress the sender's IP or hostname
 * @param {Number} remotePort the sender's port
 */
function onMessage(data, remoteAddress, remotePort) {
};

/**
 * Called when an error occurs while listening for data
 * @param {Error} error information about the error
 */
function onError(error) {
};
```

cpq, 2016-03-06:
Thank you @lukevenediger !
So in order to talk to the Firebase, the only thing required is plain UDP API?

lukevenediger, 2016-03-06:
Hi @cpq,

Firebase's protocol is closed, unfortunately, otherwise I'd suggest using websockets and talking to them directly. Having a middle-man service like the firebase-udp-bridge lets low-powered devices q

cpq, 2016-03-10:
Thank you @lukevenediger  - that makes sense. Let us do it.

lukevenediger, 2016-03-13:
Woo hoo thanks @cpq!

Bluebie, 2016-04-15:
This is awesome to see in the works. I'd like to use UDP to implement an Art-Net receiver for stage lighting, like Daft Punk styled LED helmets, which would integrate with regular DMX lighting consoles in venues to stream video on to the helmets wirelessly. It could also be used to implement things like Apple's HomeKit or the LIFX lightbulb LAN protocol.

One of the most interesting things enabled by these extremely cheap wifi devices is the idea of every little atomic thing being it's own isolated network device, like you could imagine building a robot where every motor has it's own wifi computer, and all motors can be discovered via UDP announcements in a host computer and controlled together as one unit, reconfigurable like lego. UDP's ability to do broadcasts, and importantly, transmit data without any reliability guarantees over WiFi, makes it essential for anyone building low latency distributed systems.

cpq, 2016-04-22:
@Bluebie is there a web page that describes your Art-Net and/or HomeKit projects? They sound fantastic!

lukevenediger, 2016-04-22:
I see the docs made it in today (https://github.com/cesanta/iot/commit/db4329eda7020ca2715dd18e3917406a7522c315) - not long now. Thanks for your effort!

alashkin, 2016-05-13:
We implemented Node-like TCP & UDP API
See here for details:
https://docs.cesanta.com/mongoose-iot/latest/#/javascript-api/udp/

Feel free to try and report bugs ;)

closed 2016-05-13 by alashkin

lukevenediger, 2016-05-13:
Wooo hoo! thanks @alashkin! I'll give it a try this weekend - yeah!!