cesanta/mongoose-os #101: GPIO.setisr stuck

intx82, 2016-03-09:
When use GPIO.setisr and pin goes to another state esp module halted and reseted with wdt

```

smartjs 34760/696$ GPIO.setmode(14,3,0);
false
smartjs 35304/792$ GPIO.setmode(14,1,0);
true
smartjs 35240/692$ GPIO.setisr(14,3,function(){print("GOTISR");});
true

```
after this, i changed pin state and esp-12q stuck, and reseted with wdt

PS: Может лучше по русски?

realzeus, 2016-03-22:
The same behavior on ESP-01.

alashkin, 2016-03-22:
Hi!
Try to rebuild smartjs@master, should be fixed.
```
GPIO.setmode(14,3,0);
GPIO.setisr(14,3,function(){print("GOTISR");});
```

`GPIO.setmode(14,3,0);` should return `true`, is as an issue as well.

intx82, 2016-04-05:
+alashkin
Yes, it's work!   thank you for review this issue.

alashkin, 2016-04-05:
Cool!
Thank you for reporting issue.

closed 2016-04-05 by alashkin