cesanta/mongoose-os #102: Trying to read PWM length from HC-SR04 Ultrasonic Sensor - is there a way to get the current time in Microseconds?

lukevenediger, 2016-03-16:
Hi,

I'm trying to use smartjs to read the duration value from a HC-SR04 ultrasonic distance sensor. I can set an interrupt on the Trigger pin but I don't have a way to measure the duration in microseconds. Arduino has a [micros](https://www.arduino.cc/en/Reference/Micros) function that gives you the time, in microseconds, since the board was powered on.

Is there another way for me to read PWM singal duration or to get the current time in microseconds?

Thanks,
Luke

rojer, 2016-03-16:
in 83f3b4e81f55c8e531032a9f2acafbcc02e25658 i added `Sys.time()`, will that do?

rojer, 2016-03-16:
i'm going to say it is, please reopen if this does not address your problem.

closed 2016-03-16 by rojer

lukevenediger, 2016-03-17:
hey @rojer thanks! I'll check that out and re-open if necessary.